arduino/arduino-cli #1955: Compilation fails without any error message

IvarWithoutBones, 2022-10-27:
### Describe the problem

When attempting to compile an completely empty sketch on MacOS, the CLI return exit status one with seemingly no error message. This is with a fresh install, the only thing done prior to receiving the error is installing the needed platform/core. 

```
~/projects/arduino/test > arduino-cli compile --fqbn arduino:avr:uno --verbose --clean test.ino                                                     NORMAL
Using board 'uno' from platform in folder: /Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5
Using core 'arduino' from platform in folder: /Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5
Detecting libraries used...
/Users/ivv/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/variants/standard /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/sketch/test.ino.cpp -o /dev/null
Generating function prototypes...
/Users/ivv/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/variants/standard /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/sketch/test.ino.cpp -o /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/ivv/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/preproc/ctags_target_for_gcc_minus_e.cpp

Used platform Version Path
arduino:avr   1.8.5   /Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5

Error during build: exit status 1
```

### To reproduce

```bash
#!/usr/bin/env bash
rm -rf /Users/ivv/Library/Arduino15
arduino-cli core install arduino:avr
arduino-cli sketch new test
cd test
arduino-cli compile --fqbn arduino:avr:uno --verbose --clean test.ino
```

### Expected behavior

The sketch should successfully compile and spit out a binary.

### Arduino CLI version

arduino-cli  Version: nightly-20221025 Commit: f239754 Date: 2022-10-25T01:35:53Z

### Operating system

macOS

### Operating system version

  12.6

### Additional context

Adding `--build-path $PWD/build` shows that all files that are mentioned in the log look correct, it just stops at some later point (potentially linking of preprocessed files, or executing gcc?). The only thing that seemed fishy to me is the `compile_commands.json` being completely empty.

 I would appreciate some help with debugging this, I'm a bit lost myself.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-28:
Hi @IvarWithoutBones. Thanks for your report.

What happens if you run this command from the command line?

```text
/Users/ivv/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags --help
```

---

I notice you have included the `--clean` flag in your commands. Does the problem still occur even without that flag, or is it a required condition to reproduce the fault?

IvarWithoutBones, 2022-10-28:
Thank you for the swift response, its much appreciated.

> What happens if you run this command from the command line?
> `/Users/ivv/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags --help`

It prints the help message as expected, there does not seem to be anything broken there. When invoking the executable in the same way `arduino-cli` does based on the logs I got an error claiming it cannot open a temporary file, assuming that is just because of cleanup though.

> I notice you have included the --clean flag in your commands. Does the problem still occur even without that flag, or is it a required condition to reproduce the fault?

It is not, the problem occurs exactly the same with or without the flag. I just used it to ensure no potentially corrupted cached build artifacts were used.

When adding the `--preprocess` flag a more descriptive error is thrown, which sounds its the root cause of the issue.
```bash
~/projects/arduino/test > arduino-cli compile -b arduino:avr:uno -v  --preprocess                                                                   INSERT
Using board 'uno' from platform in folder: /Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5
Using core 'arduino' from platform in folder: /Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5
/Users/ivv/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/variants/standard /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/sketch/test.ino.cpp -o /dev/null
/Users/ivv/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino -I/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/variants/standard /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/sketch/test.ino.cpp -o /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/ivv/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/tmp/arduino-sketch-6379EA0C0E6E5E5AAB9581BFB5390945/preproc/ctags_target_for_gcc_minus_e.cpp
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x1007fc8a1]

goroutine 1 [running]:
github.com/arduino/arduino-cli/commands/compile.Compile({0x1012d1600?, 0x100a48102?}, 0xc000172000, {0x100cd95a0?, 0xc000120008}, {0x100cd95a0?, 0xc000120010}, 0x0, 0x64?)
	github.com/arduino/arduino-cli/commands/compile/compile.go:198 +0x1881
github.com/arduino/arduino-cli/cli/compile.runCompileCommand(0xc00038a780?, {0xc0002e4600, 0x0, 0x4?})
	github.com/arduino/arduino-cli/cli/compile/compile.go:235 +0xa9b
github.com/spf13/cobra.(*Command).execute(0xc00038a780, {0xc0002e45c0, 0x4, 0x4})
	github.com/spf13/cobra@v1.2.1/command.go:860 +0x663
github.com/spf13/cobra.(*Command).ExecuteC(0xc0002acf00)
	github.com/spf13/cobra@v1.2.1/command.go:974 +0x3bd
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.2.1/command.go:902
main.main()
	github.com/arduino/arduino-cli/main.go:31 +0x77
```
#1947 sounds like it reports a similar segfault after not being able to find a header, although only `Arduino.h` is included which does exist in `/Users/ivv/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/`.