arduino/arduino-cli #1957: More permissive license for *.proto files

benmcmorran, 2022-10-28:
### Describe the request

The *.proto files in this repo are currently licensed under GPLv3. I'm not a lawyer, but that seems like it would prevent any non-GPL application from interacting arduino-cli over the gRPC interface because the application would need the *.proto files for code generation. Is that intentional, and if not, would it be possible to change the license for *.proto files specifically to something more permissive like MIT? It's seems strange that an MIT licensed application, for example, would be allowed to interact with arduino-cli over the CLI interface, but not the gRPC interface.

### Describe the current behavior

The *.proto files in this repo are currently licensed under GPLv3.

### Arduino CLI version

0.28.0

### Operating system

N/A

### Operating system version

N/A

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

benmcmorran, 2022-11-21:
@alranel any updates on if this change is something that Arduino is considering?

benmcmorran, 2023-11-30:
Great! Thanks for the update.

Xayton, 2024-06-13:
It was confirmed internally by @alranel that we can proceed by using the Apache-2.0 license.

closed 2024-06-27 by cmaglie

alessio-perugini, 2024-07-03:
@benmcmorran starting from [v1.0.2](https://github.com/arduino/arduino-cli/releases/tag/v1.0.2) the .proto file license is now published and shipped with the apache2 license.

PR #2653 (2024-06-27): Changed license of .protoc definition to Apache-2.0
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [X] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / feature [... cut]