arduino/arduino-cli #1959: Cannot install new board managers

angrest, 2022-11-01:
### Describe the problem

An existing github-action (not used for a month) now breaks if I try to install another platform:
```
Run arduino-cli config init
[2](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:2)
  arduino-cli config init
[3](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:3)
  arduino-cli core update-index
[4](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:4)
  arduino-cli core upgrade
[5](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:5)
  arduino-cli core install ATTinyCore:avr  --additional-urls http://drazzy.com/package_drazzy.com_index.json
[6](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:6)
  shell: /usr/bin/bash -e {0}
[7](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:7)
  env:
[8](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:8)
    REPO_PATH: libraries/pocketBME280
[9](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:10)
Config file written to: /home/runner/.arduino15/arduino-cli.yaml
[10](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:11)

[11](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:12)
Downloading index: library_index.tar.bz2 0 B / 2.15 MiB    0.00%
[12](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:13)
Downloading index: library_index.tar.bz2 downloaded
[13](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:14)

[14](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:15)
Downloading index: package_index.tar.bz2 0 B / 43.47 KiB    0.00%
[15](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:16)
Downloading index: package_index.tar.bz2 downloaded
[16](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:17)

[17](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:18)
Downloading index: package_index.tar.bz2 0 B / 43.47 KiB    0.00%
[18](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:19)
Downloading index: package_index.tar.bz2 downloaded
[19](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:20)
Downloading missing tool builtin:ctags@5.8-arduino11...
[20](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:21)

[21](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:22)
builtin:ctags@5.8-arduino11 0 B / 108.99 KiB    0.00%
[22](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:23)
builtin:ctags@5.8-arduino11 downloaded
[23](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:24)
Installing builtin:ctags@5.8-arduino11...
[24](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:25)
builtin:ctags@5.8-arduino11 installed
[25](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:26)
Downloading missing tool builtin:serial-discovery@1.3.2...
[26](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:27)

[27](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:28)
builtin:serial-discovery@1.3.2 downloaded
[28](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:29)
Installing builtin:serial-discovery@1.3.2...
[29](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:30)
builtin:serial-discovery@1.3.2 installed
[30](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:31)
Downloading missing tool builtin:mdns-discovery@1.0.6...
[31](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:32)

[32](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:33)
builtin:mdns-discovery@1.0.6 downloaded
[33](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:34)
Installing builtin:mdns-discovery@1.0.6...
[34](https://github.com/angrest/pocketBME280/actions/runs/3370241682/jobs/5590895387#step:4:35)
[... 190 lines cut ...]
### Expected behavior

board manager is installed without errors.

### Arduino CLI version

nightly-20221025 

### Operating system

Linux

### Operating system version

Linux ubuntu-latest 5.4.199-odroidxu4 #22.05.3 SMP PREEMPT Wed Jun 22 07:29:40 UTC 2022 armv7l armv7l armv7l GNU/Linux

### Additional context

This is reproducible with both the ATTTiny board manager from the example above as well as with the ESP32 board manager `https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json`

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-01:
Hi @angrest. You must run the `arduino-cli core update-index` command before `core install`.

closed 2022-11-01 by per1234

angrest, 2022-11-01:
did not solve the problem.
freshly created docker container as above
```
root@ubuntu-latest:/# arduino-cli config init
Config file written to: /root/.arduino15/arduino-cli.yaml
root@ubuntu-latest:/# arduino-cli core update-index
Downloading index: library_index.tar.bz2 downloaded                             
Downloading index: package_index.tar.bz2 downloaded                             
Downloading index: package_index.tar.bz2 downloaded                             
root@ubuntu-latest:/# arduino-cli core install esp32:esp32 --additional-urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                          
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.3.2...
builtin:serial-discovery@1.3.2 downloaded                                       
Installing builtin:serial-discovery@1.3.2...
builtin:serial-discovery@1.3.2 installed
Downloading missing tool builtin:mdns-discovery@1.0.6...
builtin:mdns-discovery@1.0.6 downloaded                                         
Installing builtin:mdns-discovery@1.0.6...
builtin:mdns-discovery@1.0.6 installed
Downloading missing tool builtin:serial-monitor@0.10.0...
builtin:serial-monitor@0.10.0 downloaded                                        
Installing builtin:serial-monitor@0.10.0...
builtin:serial-monitor@0.10.0 installed
Error initializing instance: Loading index file: loading json index file /root/.arduino15/package_esp32_index.json: open /root/.arduino15/package_esp32_index.json: no such file or directory
Error initializing instance: Loading index file: loading json index file /root/.arduino15/package_esp32_index.json: open /root/.arduino15/package_esp32_index.json: no such file or directory
Invalid argument passed: Found 0 platform for reference "esp32:esp32":
```

per1234, 2022-11-01:
If you are going to specify package index URLs via the `--additional-urls` flag, you must do that in every command that relies on that information. `core update-index` very much relies on that information.

If you need additional assistance using Arduino CLI, we are always happy to help out over on its dedicated category in the Arduino forum:

https://forum.arduino.cc/c/software/arduino-cli/89

angrest, 2022-11-01:
as written above, the action worked without any problem four weeks ago (it is actually based on your contribution 😊 ) - so, why should it stop working suddenly? The only thing which changed meanwhile is arduino-cli was updated to 0.28.0

angrest, 2022-11-02:
I've meanwhile tried v0.27.0 - there it works without problem (with and without previous `core update-index`). 
Fresh container:
```
root@ubuntu-latest:/# curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s 0.27.0
Installing in //bin
ARCH=ARMv7
OS=Linux
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.27.0_Linux_ARMv7.tar.gz

A new release of Arduino CLI is available: 0.27.0 → 0.28.0
https://arduino.github.io/arduino-cli/latest/installation/#latest-packages
arduino-cli  Version: 0.27.0 Commit: c2af7c5a Date: 2022-09-05T08:10:30Z installed successfully in //bin
root@ubuntu-latest:/# arduino-cli config init
Config file written to: /root/.arduino15/arduino-cli.yaml
root@ubuntu-latest:/# arduino-cli core install esp32:esp32 --additional-urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
Downloading index: library_index.json.gz downloaded                             
Downloading index signature: library_index.json.sig downloaded                  
<output truncated>
Platform esp32:esp32@2.0.5 installed
root@ubuntu-latest:/#