arduino/arduino-cli #1963: Provide an `--offline` directive to prevent errors in portable mode

ubidefeo, 2022-11-04:
### Describe the request

I have been testing a setup for portable mode of Arduino CLI.
I am happy to know that (at least on Mac OS) it works beautifully, but when I have an unrecognised device attached to the USB port, it tries to resolve the `fqbn` using the Arduino online builder.
This generates an error which is unpleasant.
I am not sure what other calls to services are made besides updating index for libraries and platforms, but in theory a user might decide to manually copy the platforms and related index files into a portable directory and run `arduino-cli` from there

### Describe the current behavior

with the following file structure, `arduino-cli` can be run locally and off removable storage.
The requirements are that any platform and its supporting package index can be found by the CLI.
A local `arduino-cli.yaml` is required and must be properly customised (follows)

[image]

I have manually copied what was needed from my `Arduino15` folder and defined user and data paths for the configuration `.yaml` file

**arduino-cli.yaml**
```
board_manager:
  additional_urls:
  - https://dl.espressif.com/dl/package_esp32_index.json
daemon:
  port: "50051"
directories:
  data: ./Arduino15
  downloads: ./Arduino15/staging
  user: ./Arduino
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
output:
  no_color: false
sketch:
  always_export_binaries: true
updater:
  enable_notification: true
```

### Arduino CLI version

test-1845-3fd2864ff8d77d476d1e73500f8a0c54d03354b3-git-snapshot

### Operating system

macOS

### Operating system version

12.6

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

cmaglie, 2022-11-04:
> This generates an error which is unpleasant.

May you paste the error too?

matthijskooijman, 2022-11-04:
I'm also not sure I understand the relation here. Why would a portable installation not be able to do an online lookup? Or is the problem that a portable installation has no place to download index files or so? If so, an "offline" flag does not seem like an appropriate fix. Even more, if this error always occurs in portable mode, why need a flag at all, couldn't it just be applied always in portable mode?

cmaglie, 2022-11-08:
The "portable mode" @ubidefeo is referring to is just changing the "data" directory to a path inside the arduino-cli installation, so there is no way to tell if we are in "portable" or not.

> Why would a portable installation not be able to do an online lookup? 
I agree with you, the whole reasoning about the "portable mode" is useless in this issue and I think that the error is present even in a normal installation.

ubidefeo, 2022-11-17:
I'll better explain the reason for such request, and close the issue because this can be addressed in a different way: graceful handling of errors, without blocking the rest of the execution.
In this comment one of the problems comes through
https://github.com/arduino/arduino-cli/issues/880#issuecomment-1313623787

we have two main needs:
* being able to maintain JSON formatting
* not stopping execution of a flow when an unexpected error happens

For the portable case, assume the following scenario:
A user has N package index files in their portable configuration (see file tree screenshot in this issue).
If a command such as `./arduino-cli board list --format json` is issued, and the `VID:PID` cannot be mapped to something defined in one of the `boards.txt` entries, the CLI calls Arduino Builder to receive back a possible match.
If the service is not reachable (offline) then the error is generated and no valid JSON is output.

Because these should be addressed in #1982 , I'll close this issue and call it a day

PR #1982 (2022-11-21): Improve `board list` detection via cloud API
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [X] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [  [... cut]
closed 2022-11-17 by ubidefeo