arduino/arduino-cli #1968: The CLI provides different error messages when loading the main sketch file fails

kittaakos, 2022-11-08:
### Describe the problem

IDE2 must gracefully handle invalid sketch (folder) names and open them even if the CLI cannot load the sketch.

Since CLI can't provide error codes (https://github.com/arduino/arduino-cli/issues/1762), IDE2 must parse the error message of the CLI and try to detect invalid sketch name problems. (See arduino/arduino-ide#1563)

Unfortunately, the CLI's error message quite often varies. As a consumer of the CLI, I would expect that the error message is the same if the problem is the same.

### To reproduce

```
% tree                                                                        
.
├── Bar
│   └── xxx.ino
└── foo
    └── Foo.ino

2 directories, 2 files
% ~/dev/git/arduino-ide/arduino-ide-extension/build/arduino-cli version       
arduino-cli  Version: 0.28.0 Commit: 06fb1909 Date: 2022-10-20T08:42:20Z
% ~/dev/git/arduino-ide/arduino-ide-extension/build/arduino-cli compile -b arduino:avr:uno ./Bar 
Error opening sketch: main file missing from sketch: /Users/a.kitta/Desktop/invalid-sketches/Bar/Bar.ino
% ~/dev/git/arduino-ide/arduino-ide-extension/build/arduino-cli compile -b arduino:avr:uno ./foo 
Error opening sketch: no valid sketch found in /Users/a.kitta/Desktop/invalid-sketches/foo: missing /Users/a.kitta/Desktop/invalid-sketches/foo/foo.ino
```

```
Error opening sketch: main file missing from sketch: /Users/a.kitta/Desktop/invalid-sketches/Bar/Bar.ino
```

```
Error opening sketch: no valid sketch found in /Users/a.kitta/Desktop/invalid-sketches/foo: missing 
```

### Expected behavior

No matter the final error message, it's the same if I want to load, compile, etc an invalid sketch. It should no matter if it's `/foo/Foo.ino` or `/Bar/xxx.ino` invalid, it's just invalid.

### Arduino CLI version

Version: 0.28.0 Commit: 06fb1909 Date: 2022-10-20T08:42:20Z

### Operating system

macOS

### Operating system version

12.5.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-11-08:
> IDE2 must parse the error message of the CLI

Too bad, this won't work at all. IDE2 cannot rely on parsing the error message as it's different per locale. After setting the `locale` to `it` in the `arduino-cli.yaml` config file.

```
% ~/dev/git/arduino-ide/arduino-ide-extension/build/arduino-cli compile -b arduino:avr:uno ./Bar --config-file ~/.arduinoIDE/arduino-cli.yaml
Si è verificato un errore durante l'apertura dello sketch: il file principale manca dallo sketch: /Users/a.kitta/Desktop/invalid-sketches/Bar/Bar.ino
```

```
% ~/dev/git/arduino-ide/arduino-ide-extension/build/arduino-cli compile -b arduino:avr:uno ./foo --config-file ~/.arduinoIDE/arduino-cli.yaml
Si è verificato un errore durante l'apertura dello sketch: Non è stato trovato uno sketch valido in /Users/a.kitta/Desktop/invalid-sketches/foo: manca /Users/a.kitta/Desktop/invalid-sketches/foo/foo.ino
```

Bikappa, 2023-01-03:
The two cases are logically different, at least from the code perspective. In the [sketch initializer](https://github.com/arduino/arduino-cli/blob/558130b1b4a2f67895b5ec88b97c5884227d78ef/arduino/sketch/sketch.go#L64) we look for a file named as the directory with the `.ino` extension as possible main file candidate.
- with `compile ./foo` we find the file `./foo/Foo.ino` (on MacOS and probably Windows as the locations are case insensitive there), which is then eventually considered invalid, when empty for example.
- with `compile ./Bar` no file is found as main file candidate.

So the source problems seem to be the different and might deserve a different error message. @kittaakos wdyt?

cmaglie, 2023-01-03:
> which is then eventually considered invalid, when empty for example.

actually, it is explicitly checked for the correct file case https://github.com/arduino/arduino-cli/blob/558130b1b4a2f67895b5ec88b97c5884227d78ef/arduino/sketch/sketch.go#L192-L226

Maybe we can "normalize" the error message with `"Invalid sketch found: <REASON>"` so the check on the IDE side can be done against the prefix `"Invalid sketch found:"`.

BTW as @kittaakos suggested this is fragile because the messages are localized, the real solution is to implement error codes #1762

kittaakos, 2023-01-06:
> The two cases are logically different, at least from the code perspective.

Thank you for explaining this, but I opened the issue from the user's perspective.

> So the source problems seem to be the different and might deserve a different error message.

I am not sure. As a user, I do know/mind how this is working internally. I want to load two sketches; both of them have an invalid name according to the [spec](https://arduino.github.io/arduino-cli/latest/sketch-specification/#primary-sketch-file), so I would expect the same error message. The detail/cause of the errors if any can differ.

> Every sketch must contain a `.ino` file with a file name matching the sketch root folder name.

> BTW as @kittaakos suggested this is fragile because the messages are localized, the real solution is to implement error codes #1762

I agree. Probably, no need to do anything with this issue once #1762 is resolved.