arduino/arduino-cli #1970: Sometimes it will not compile after switching development version

Tinyu-Zhao, 2022-11-09:
### Describe the problem

Sometimes it fails to compile after switching to the development version, and it says "No connection established".

[image]

But strangely enough, after restarting the IDE, most of them will be able to compile normally.

### To reproduce

Sometimes switching clients and compiling will cause this problem, and sometimes not.

### Expected behavior

2.0.1

### Arduino IDE version

Normal compilation

### Operating system

macOS

### Operating system version

Big Sur 11.7 (20G817)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-11:
Hi all. Thanks for your reports. [**cmaglie**](https://github.com/cmaglie) investigated the problem and identified a potential cause.

[A fix has been prepared](https://github.com/arduino/arduino-cli/pull/1972), but unfortunately we were never able to reproduce the fault despite quite some attempts. So we are hoping that you might be willing to help us out by testing the fix and letting us know whether the error still occurs.

Everyone is welcome to test, but the most conclusive results will be in the case where the error is produced reliably with the regular production version of the software. So please first test with Arduino IDE 2.0.1 (or the [latest nightly build of Arduino IDE](https://www.arduino.cc/en/software#nightly-builds) if you prefer) to make sure you are able to reproduce the error.

I'll provide the instructions for getting the tester build of Arduino IDE that has the proposed fix:

1. Sign in to your [**GitHub**](https://github.com/) account.
   (GitHub only allows downloads of the tester builds when you are signed in.)
1. Open this page in your browser:
   <https://github.com/arduino/arduino-ide/actions/runs/3444944271#artifacts>
1. From the "**Artifacts**" section of the page, click the download link for your operating system.
   **ⓘ** For example, if you are using Windows, use the "**Windows_X86-64_zip**" link.
1. Wait for the download to finish.
1. Extract or install the downloaded file as usual.
1. Start the tester build of Arduino IDE you downloaded.
1. Perform whatever operations normally produce the error when using the regular version of Arduino IDE.

After doing that, please comment here to let me know the following:

- Were you able to reproduce the fault using the regular version of Arduino IDE?
- Were you able to reproduce the fault using the tester version of Arduino IDE?

---

Please let me know if you have any questions or problems while following those instructions.

ReneHeuker, 2022-11-11:
Had same problem on Mac with the ide for arduino uno and esp01. problem occurs on regular version and disappears after installing this snapshot.

simap, 2022-11-11:
Had the same problem after updating to 2.0.1 w/ esp32, went away when I restarted the IDE. 🤷‍♂️

Burt-Silverman, 2022-11-13:
>Were you able to reproduce the fault using the regular version of Arduino IDE?
Yes, I am using ESP8266 rather than ESP32 for my work.
>Were you able to reproduce the fault using the tester version of Arduino IDE?
No, the tester version works fine for me.

Andrew-F-in-Australia, 2022-11-13:
logs below.  Test version (beta version) worked perfectly when executed through a terminal.  Crashed when run on a clean install of Fedora 36 when executed by double clicking in terminal/graphical display manager.

Logs here: https://forum.arduino.cc/t/linux-nano-ch340-detected-working-in-ide1-8-13-but-not-ide2-0-1/1048313/15

The current fix also allows the user to select the example sketches.  See detailed logs.  Running *very* tired when testing (and typing this) so may have made an error my side.  I'll check again with fresh eyes tomorrow.  

Let me know if you want another beta version tested later please if you can see a simple fix.  The error code when this is double-clicked seemed logical, but... ... ...

limited, 2022-11-13:
* I was able to reproduce the fault with the regular Arduino IDE version (stack trace below)
* I was not able to reproduce the fault with the tester version.

```
Checking for update
Update for version 2.0.1 is not available (latest version: 2.0.1, downgrade is allowed).
root ERROR ---- !!!SLOW!!! DURATION: search took 1089.935 ms. Args: [{"type":"Updatable"}] ----
root WARN Timeout after 5000 millis. The editor has not shown up in time. URI: file:///private/var/folders/6k/rm3xpqmn4sg737d737mscccw0000gq/T/.arduinoIDE-unsaved20221013-90783-199s8mm.at44/sketch_nov13a/sketch_nov13a.ino
daemon INFO panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x42ffc5d]

goroutine 12 [running]:
github.com/arduino/go-properties-orderedmap.(*Map).Merge(0x0, {0xc00550b880, 0x1, 0x0})
	/go/pkg/mod/github.com/arduino/go-properties-orderedmap@v1.6.0/properties.go:443 +0x3d
github.com/arduino/go-properties-orderedmap.(*Map).Clone(...)
	/go/pkg/mod/github.com/arduino/go-properties-orderedmap@v1.6.0/properties.go:477
github.com/arduino/arduino-cli/arduino/cores.(*Board).GetBuildProperties(0xc0001a4120, 0xc008166a20)
	/home/build/arduino/cores/board.go:119 +0xd5
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*Explorer).ResolveFQBN(0xc0053c6200, 0xc0053a8140)
	/home/build/arduino/cores/packagemanager/package_manager.go:291 +0x105
github.com/arduino/arduino-cli/commands/upload.SupportedUserFields({0x400cd45, 0x49b9100}, 0xc0070d2000)
	/home/build/commands/upload/upload.go:62 +0x159
github.com/arduino/arduino-cli/commands/daemon.(*ArduinoCoreServerImpl).SupportedUserFields(0x4967f60, {0x4c5e870, 0xc008166000}, 0x0)
	/home/build/commands/daemon/daemon.go:369 +0x2c
github.com/arduino/arduino-cli/rpc/cc/arduino/cli/commands/v1._ArduinoCoreService_SupportedUserFields_Handler({0x49cdd60, 0xc0003c0d00}, {0x4c5e870, 0xc008166000}, 0xc00283c000, 0x0)
	/home/build/rpc/cc/arduino/cli/commands/v1/commands_grpc.pb.go:1775 +0x173
google.golang.org/grpc.(*Server).processUnaryRPC(0xc0003408c0, {0x4c70c18, 0xc000296f00}, 0xc00267a000, 0xc0004ccd80, 0x52b2748, 0x0)
	/go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:1286 +0xc8f
google.golang.org/grpc.(*Server).handleStream(0xc0003408c0, {0x4c70c18, 0xc000296f00}, 0xc00267a000, 0x0)
	/go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:1609 +0xa2a
google.golang.org/grpc.(*Server).serveStreams.func1.2()
	/go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:934 +0x98
created by google.golang.org/grpc.(*Server).serveStreams.func1
	/go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:932 +0x294
```

closed 2022-11-14 by cmaglie

Andrew-F-in-Australia, 2022-11-14:
Hi all,

I had one of the forum issues above.

Confirming....
Nightly build 2.0.2 worked perfectly in a fresh build of Fedora, if you double click on the icon in file manager.
To get appimage working successfully in Fedora/Linux.

(1) sudo usermod -aG dialout horace
(2) sudo chmod +x arduino-ied_2.0.1_Linux_64bit.appimage {or whatever filename it is.]
(3) sudo chown <username>: /dev/ttyUSB0 (or whatever port you are connected to.
(4) reboot, or at the least, log out and log back in again.

Then open app image by double clicking in graphical file manager.

Functions perfectly this way.

--------------------------------------------
Had one successful run and one daemon panic in command line.

running in terminal/command line through ./arduino-ide_nightly-20221114_Linux_64bit.AppImage gives the following kernel panic

`daemon INFO panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x6fc2bd]
`
[ctrl]-C in the terminal to force-close the program, then opening through ./arduino-ide_nightly-20221114_Linux_64bit.AppImage gave a perfectly operating IDE the second time around.

Might be a linux bug we live with for a while - it's intermittent, so a re-loading of the program appears to fix it.

Thanks again to you and Cristian.

Regards,

A

TheAeroPath, 2022-11-15:
> I encounter same problem today, without using above fix, I just uninstall esp32 from board manager and re-install, the problem fixed

Fixed the problem for me as well. Thank you much!

emrullahjack, 2022-11-17:
As a temporary solution, I re-install my board manager everytime I restart my computer to fix it.

The problem appears to occur when I click Tools > board > _board_manager_ 
`daemon INFO panic: runtime error: invalid memory address or nil pointer dereference`
After that point it seems that the connection between the CLI and the GUI is broken

per1234, 2022-11-17:
Hi @emrullahjack. Please give the new Arduino IDE 2.0.2 release a try. The bug should be fixed in that version.

The download links are listed here:

https://www.arduino.cc/en/software

emrullahjack, 2022-11-17:
Hi @per1234 thank you! 2.0.2 release works perfectly fine for me!!