arduino/arduino-cli #1981: Redundant discovery processes created by additional daemon instances

per1234, 2022-11-17:
### Describe the problem

[Pluggable discovery](https://arduino.github.io/arduino-cli/dev/pluggable-discovery-specification/) tools supply Arduino CLI with a list of "ports" of connected Arduino boards. When Arduino CLI is [running as a daemon](https://arduino.github.io/arduino-cli/dev/commands/arduino-cli_daemon/), these tools are started when a gRPC request is received that requires a port listing and then left running for the remainder of the session.

🐛 Redundant processes for each discovery tool are created by additional instances of the Arduino CLI daemon. It is possible that discovery tools may not be designed to operate correctly in this way, and the multiple processes might interfere with each other.

### To reproduce

#### Setup

```text
$ arduino-cli version

arduino-cli  Version: 0.29.0-rc.1 Commit: 76251df9 Date: 2022-11-17T03:35:12Z

$ arduino-cli daemon

Daemon is now listening on 127.0.0.1:50051
```

#### Demo

run the following [`grpcurl`](https://github.com/fullstorydev/grpcurl) commands in another terminal:

```text
$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Create

{
  "instance": {
    "id": 1
  }
}

$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 1}}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Init

$ pgrep -a -f 'serial-discovery'  # Discovery tools have not yet been started

$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 1}}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.BoardList

[...]

$ pgrep -a -f 'serial-discovery'  # A process for each discovery tool was created

134825 /home/per/.arduino15/packages/builtin/tools/serial-discovery/1.3.2/serial-discovery

$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Create

{
  "instance": {
    "id": 2
  }
}

$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 2}}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Init

$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 2}}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.BoardList

[...]

$ pgrep -a -f 'serial-discovery'

134825 /home/per/.arduino15/packages/builtin/tools/serial-discovery/1.3.2/serial-discovery
135007 /home/per/.arduino15/packages/builtin/tools/serial-discovery/1.3.2/serial-discovery
```

🐛 A new process is created for each discovery tool by additional Arduino CLI daemon instances.

### Expected behavior

A single instance of each discovery tool should be shared by all Arduino CLI daemon instances.

### Arduino CLI version

76251df9

### Operating system

Windows, Ubuntu

### Operating system version

Windows 10, Ubuntu 20.04

### Additional context

Related: https://github.com/arduino/arduino-cli/issues/1960

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2023-02-01:
Fixing this requires a medium refactoring of the `DiscoveryManager`, there is an actual real-world scenario when this could be a problem? If this is not a critical issue, I'd like to postpone it.

per1234, 2025-08-12:
>there is an actual real-world scenario when this could be a problem?

Not that I am aware of. I was motivated to create this issue by the concern Paul Stoffregen expressed here:

https://forum.pjrc.com/index.php?threads/teensy4-1-reading-from-serial-monitor-of-arduino-ide.71370/page-2#post-315184

> Especially on Windows, access to many types of system resources is done with exclusive locking access, where keeping the old process running could prevent the new process from gaining access to the Windows system resources it needs to discover connected hardware.

The concern was expressed in the context of two different versions of a pluggable discovery tool being left running due to a bug: https://github.com/arduino/arduino-cli/issues/1960. However, my sense was that the concern was not fully addressed by fixing that bug alone, since we still have a separate condition under which redundant discovery processes can be created. Even though in this case they are multiple processes of the same version of the discovery rather than two different versions as was caused by https://github.com/arduino/arduino-cli/issues/1960, it seems the difference isn't a significant as it applies to the specific concern re: the processes interfering with one another.

But the concern about multiple discovery processes raised by Paul was hypothetical rather than based on a real world problem:

> Leaving the old one running is probably ok for Teensy. But maybe not such a great design?

PR #2985 (2025-08-27): fix: Multiple gRPC `Create/Init/Destroy` cycles causes multiple pluggable-discovery zombie process
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [X] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs hav [... cut]