arduino/arduino-cli #1990: arduino-cli board list throws bad CPU type in executable on a M2 Mac without rosetta emulation installed

aleroot, 2022-11-23:
### Describe the problem

I have an M2 MacBookPro and I have installed arduino-cli_0.29.0_macOS_ARM64, when I try to list the board I get 2 : bad CPU type in executable ...

I don't want to install rosetta, so is there a way to have all the underlying tools compiled for ARM64?

What is the sense of having arduino-cli ARM64 if then uses underlying tools that are not ARM64 ?

### To reproduce

install arduino-cli ARM64 on an Apple silicon Mac that doesn't have installed Rosetta, then run `./arduino-cli board list`

### Expected behavior

Everything should work without giving errors: bad CPU type in executable

### Arduino CLI version

arduino-cli  Version: 0.29.0 Commit: 76251df9 Date: 2022-11-17T09:21:37Z

### Operating system

macOS

### Operating system version

Ventura 13.0.1 (22A400)

### Additional context

Output of arduino-cli board list

```
alessio@AleMacBookPro Tools % ./arduino-cli board list
Downloading index: library_index.tar.bz2 downloaded                                                                                                                                                         
Downloading index: package_index.tar.bz2 downloaded                                                                                                                                                         
Downloading missing tool builtin:serial-monitor@0.10.0...
builtin:serial-monitor@0.10.0 downloaded                                                                                                                                                                    
Installing builtin:serial-monitor@0.10.0...
builtin:serial-monitor@0.10.0 installed
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                                                                                                      
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.3.2...
builtin:serial-discovery@1.3.2 downloaded                                                                                                                                                                   
Installing builtin:serial-discovery@1.3.2...
builtin:serial-discovery@1.3.2 installed
Downloading missing tool builtin:mdns-discovery@1.0.6...
builtin:mdns-discovery@1.0.6 downloaded                                                                                                                                                                     
Installing builtin:mdns-discovery@1.0.6...
builtin:mdns-discovery@1.0.6 installed
Error starting discovery: discovery builtin:mdns-discovery process not started: fork/exec /Users/alessio/Library/Arduino15/packages/builtin/tools/mdns-discovery/1.0.6/mdns-discovery: bad CPU type in executable
Error starting discovery: discovery builtin:serial-discovery process not started: fork/exec /Users/alessio/Library/Arduino15/packages/builtin/tools/serial-discovery/1.3.2/serial-discovery: bad CPU type in executable
No boards found.
```

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

umbynos, 2023-01-18:
> What is the sense of having arduino-cli ARM64 if then uses underlying tools that are not ARM64 ?

Unfortunately, there are a lot of tools to support in order to allow operation like the compile.
But I'm in favour of at least provide at least `mdns-discovery` and `serial-discovery`, so that at least the basic operation would work, like listing the boards

per1234, 2023-01-18:
Since native Apple Silicon builds of Arduino CLI are already produced, and Arduino CLI's tool installation system will already install the native Apple Silicon builds of tools when those are available (falling back to installing x86 when they are not), I'll close this as off topic. 

We are now tracking the requests for Apple Silicon builds of **serial-discovery** and **mdns-discovery** in their own repositories:

- https://github.com/arduino/serial-discovery/issues/55
- https://github.com/arduino/mdns-discovery/issues/40

closed 2023-01-18 by per1234