arduino/arduino-cli #1992: Equals sign in command breaks --libraries flag

GKStretton, 2022-11-26:
### Describe the problem

The following command fails to build because it can't see the libraries installed

`arduino-cli compile --fqbn arduino:avr:mega --libraries=~/Arduino/libraries --output-dir ./build .`

The following command builds correctly

`arduino-cli compile --fqbn arduino:avr:mega --libraries ~/Arduino/libraries --output-dir ./build .`

Note the only difference is the equals sign

### To reproduce

Run 

`arduino-cli compile --fqbn arduino:avr:mega --libraries=~/Arduino/libraries --output-dir ./build .`

Or similar for any sketch that includes a library in the Arduino/libraries folder. Then change the equals sign to a space for it to work

### Expected behavior

I would expect it to work with the equals sign, equals sign is usually supported by commandline tools

### Arduino CLI version

0.29.0

### Operating system

Linux

### Operating system version

Debian 11 x86_64, kernel 5.10.0-19-amd64

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

lluiscampos, 2023-03-28:
The problem is not the `=` sign itself, but the fact that with it, the `~` symbol does not get expanded to your home directory. For example if you run with `--libraries=/home/something/Arduino/libraries` or `--libraries=$HOME/Arduino/libraries` it should work just fine.

I would like to help fixing this issue but I don't really know at which level it should be fixed. Is this something to be fixed when parsing the arguments with cobra or later on in `LibrariesLoader`? Or is it a cobra issue? (it seems related to https://github.com/spf13/cobra/issues/1577)

If someone from Arduino has some guidance I'd be happy to contribute  :smiley:

umbynos, 2023-05-29:
We tried this:
``` bash
$ echo --libraries=~/Arduino
--libraries=~/Arduino
$ echo --libraries ~/Arduino
--libraries /home/megabug/Arduino
```
we used bash on linux.

closed 2023-05-29 by umbynos

lluiscampos, 2023-05-30:
Exactly! I agree with closing this issue.

According to [bash docs](https://www.gnu.org/software/bash/manual/html_node/Tilde-Expansion.html) the tilde only gets expanded if it is a prefix. So when bash sees a token like `something=~something` will not expand it.

GKStretton, 2023-05-30:
Ah, that makes sense - thanks for explaining!