arduino/arduino-cli #1995: Save build properties in profiles

alranel, 2022-11-29:
### Describe the request

Extend the build profiles feature to also include build properties in the profiles.

Suppose you're compiling a sketch for ESP8266. In the IDE, you'll select multiple options from the Tools menu (like Flash Size, Erase Flash etc.) which affect the compilation of the core. In the CLI, unfortunately, the menu structure is not exposed so instead of using those shortcuts you need to specify all the single properties. For instance:

```
arduino-cli compile --fqbn esp8266:esp8266:generic -e --build-property build.flash_size=2M --build-property build.flash_ld=eagle.flash.2m.ld --build-property build.spiffs_pagesize=256 --build-property build.rfcal_addr=0x1FC000
```

Now, when adding `--dump-profile` I would expect such build properties to be included in the resulting file in order to get a "reproducibile build". However, they are not included.

TL;DR: build properties should be added to the sketch file specification and then implemented in `--dump-profile` and `--profile`.

### Describe the current behavior

As of now, the `--dump-profile` option implemented as part of the "reproducible builds" project (see [RFC](https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0003-build-profiles.md)) includes information about board, core and libraries but does not include any additional information such as build properties.

### Arduino CLI version

0.27.1

### Operating system

macOS

### Operating system version

10.15.7

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-11-30:
Hi @alranel. Thanks for your suggestion.

>the menu structure is not exposed so instead of using those shortcuts you need to specify all the single properties

Even though the requested capability of being able to define properties in [profiles](https://arduino.github.io/arduino-cli/dev/sketch-project-file/#build-profiles) does have valid advanced use cases (e.g., https://github.com/arduino/tooling-rfcs/issues/9), in this specific case it is not necessary and I would even strongly discourage this approach when using Arduino CLI from the command line since these properties are not intended to be part of the public interface of the platform and might be changed at any time without the platform developers considering it a breaking change.

The correct approach to setting ["custom board options"](https://arduino.github.io/arduino-cli/dev/platform-specification/#custom-board-options) is via the FQBN.

The format of the FQBN is like this:

```text
<vendor ID>:<architecture>:<board ID>[:<menu ID>=<option ID>[,<menu ID>=<option ID>]...]
```

You can learn all the available menu IDs and option IDs for a given board by running [the following `arduino-cli` command](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_board_details/):

```text
arduino-cli board details --fqbn <FQBN>
```

(where `<FQBN>` is replaced by the fully qualified board name of the board you are using)

For example:

```text
$ arduino-cli board details -b esp8266:esp8266:generic
Board name:                Generic ESP8266 Module
FQBN:                      esp8266:esp8266:generic
Board version:             3.0.2

Identification properties: board=generic

Package name:              esp8266
Package maintainer:        ESP8266 Community
Package URL:               https://arduino.esp8266.com/stable/package_esp8266com_index.json
Package website:           https://github.com/esp8266/Arduino
Package online help:       https://esp8266.com/arduino

Platform name:             esp8266
Platform category:         ESP8266
Platform architecture:     esp8266
Platform URL:              https://github.com/esp8266/Arduino/releases/download/3.0.2/esp8266-3.0.2.zip
Platform file name:        esp8266-3.0.2.zip
Platform size (bytes):     38204730
Platform checksum:         SHA-256:653add97df0a850ebc26f105355be749fb83c2db7b1ddf3e77950e694059eb9c

Required tool:             esp8266:xtensa-lx106-elf-gcc                                       3.0.4-gcc10.3-1757bed

Required tool:             esp8266:mkspiffs                                                   3.0.4-gcc10.3-1757bed

Required tool:             esp8266:mklittlefs                                                 3.0.4-gcc10.3-1757bed

Required tool:             esp8266:python3                                                    3.7.2-post1

Option:                    CPU Frequency                                                                            xtal

                           80 MHz                                                             ✔                     xtal=80

                           160 MHz                                                                                  xtal=160

Option:                    VTables                                                                                  vt

                           Flash                                                              ✔                     vt=flash

                           Heap                                                                                     vt=heap

                           IRAM                                                                                     vt=iram

Option:                    C++ Exceptions                                                                           exception

[... 281 lines cut ...]

```text
esp8266:esp8266:generic:eesz=2M
```

Much easier than digging up all those build properties!

Setting custom build options via the FQBN is covered in the platform specification (though not in a very user friendly manner) and to some extent in [the FAQ](https://arduino.github.io/arduino-cli/dev/FAQ/). I do think the documentation of this needs to be expanded though as it is a common source of confusion. We are using this public issue to track that: https://github.com/arduino/arduino-cli/issues/355

> implemented in `--dump-profile`

`--dump-profile` will configure the profile according to the FQBN passed to the command:

```text
$ ./arduino-cli compile --dump-profile --fqbn esp8266:esp8266:generic:eesz=2M

[...]

profiles:
  generic:
    fqbn: esp8266:esp8266:generic:eesz=2M
    platforms:
      - platform: esp8266:esp8266 (3.0.2)
        platform_index_url: https://arduino.esp8266.com/stable/package_esp8266com_index.json
```

alranel, 2022-11-30:
Thank you for the answer Per! That was not easy to understand from documentation indeed, and might also be a good idea to mention this in the examples related to `--dump-profile` to avoid headaches to the ESP* users that rely very much on these flags. For now, this issue and your reply will hopefully help people looking for this.
Feel free to close this issue then!

umbynos, 2023-05-29:
Closing as this is already tracked by #355

closed 2023-05-29 by umbynos