arduino/arduino-cli #2015: Avoid unnecessary compiling when clicking "Upload" button

wh201906, 2022-12-18:
### Describe the request

If no source file is changed since last uploading, there is no need to build the sketch again. The Arduino IDE can just re-use the firmware from the last build.
Users can still start a build by clicking "Verify" button.

### Describe the current behavior

When clicking "Upload" button, the Arduino IDE will start a new build even if all source files are not changed.

### Arduino IDE version

Version: 2.0.3 Date: 2022-12-05T09:30:25.331Z CLI Version: 0.29.0 [76251df9]  Copyright © 2022 Arduino SA

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-12-28:
Closing as duplicate of https://github.com/arduino/arduino-cli/issues/1996

closed 2022-12-28 by per1234

apos, 2023-06-10:
THIS IS SO ANNOYING. Really. Together with the exhaustive compilation times, not just on Windows, also on Linux for V. 2x - this is not acceptable. 

Wherever I can, I use now PlatformIO. Unfortunately, this is not possible with every Arduino project. 

This problem persists now a long time. Sorry. Can not be accepted.