arduino/arduino-cli #2021: Improve exit codes

dankeboy36, 2023-01-03:
### Describe the request

Please provide a broader range of exit codes. It's very often 1, making it hard to handle errors programmatically.

These are different kinds of programmer and user errors, but it takes time to figure them out because the exit code is the same:

 - Sketch is missing.
 - There are compiler errors.
 - The required platform is not installed.
 - Incorrect FQBN was given.
 - No FQBN was given.

### Describe the current behavior

```
XXX@ arduino-cli % cat ./test_sketch/ok/ok.ino                                  
void setup() {}
void loop() {}
XXX@ arduino-cli % cat ./test_sketch/broken/broken.ino                          
trash
XXX@ arduino-cli % ./arduino-cli version                                        
arduino-cli  Version: git-snapshot Commit: 357d4653 Date: 2023-01-03T09:30:23Z
XXX@ arduino-cli % ./arduino-cli compile -b arduino:avr:uno ./test_sketch/broken           
/Users/XXX/dev/arduino-cli/test_sketch/broken/broken.ino:1:1: error: 'trash' does not name a type; did you mean 'tanh'?
 trash
 ^~~~~
 tanh

Used platform Version Path                                                                   
arduino:avr   1.8.5   /Users/XXX/Library/Arduino15/packages/arduino/hardware/avr/1.8.5

Error during build: exit status 1
XXX@ arduino-cli % echo $?                                                                 
1
XXX@ ./arduino-cli compile -b arduino:avr:uno ./test_sketch/missing          
Error opening sketch: no such file or directory: /Users/XXX/dev/arduino-cli/test_sketch/missing
XXX@ arduino-cli % echo $?                                                     
1
XXX@ arduino-cli % ./arduino-cli compile -b arduino:mbed:nanorp2040connect ./test_sketch/ok

Used platform Version Path
                          

Error during build: Platform 'arduino:mbed' not found: platform not installed
Try running `arduino-cli core install arduino:mbed`
XXX@ arduino-cli % echo $?                                                                 
1
XXX@ arduino-cli % ./arduino-cli compile -b trash:fqbn ./test_sketch/ok                    

Used platform Version Path
                          

Error during build: Invalid FQBN: not an FQBN: trash:fqbn
XXX@ arduino-cli % echo $?                                                                 
1
XXX@ arduino-cli % ./arduino-cli compile ./test_sketch/ok 
Missing FQBN (Fully Qualified Board Name)
XXX@ arduino-cli % echo $?                               
1
XXX@ arduino-cli % 

```

### Arduino CLI version

arduino-cli  Version: git-snapshot Commit: 357d4653 Date: 2023-01-03T09:30:23Z

### Operating system

macOS

### Operating system version

11.6.5

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

scottchiefbaker, 2023-02-22:
This gets a +1 from me.

alessio-perugini, 2023-09-19:
Closed. 
Reason: https://github.com/arduino/arduino-cli/pull/2221#issuecomment-1724944029