arduino/arduino-cli #2024: Allow identification of a specific instance of a board from a Pluggable Discovery

cmaglie, 2023-01-04:
### Describe the request

We can use the USB **serial number** to know that a board has changed COM port, for example after an upload, or when it swaps to bootloader mode after a double-tap on the reset button.

Currently the discovery returns this information:

```json
{
  "eventType": "list",
  "ports": [
    {
      "address": "/dev/ttyACM0",
      "label": "/dev/ttyACM0",
      "properties": {
        "pid": "0x804e",
        "vid": "0x2341",
        "serialNumber": "EBEABFD6514D32364E202020FF10181E"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
  ]
}
```

we may extend the result by adding an extra field `hardwareId` that is populated when we have an hardware identification number available, for example for USB serial ports we can use the USB Serial Number:

```json
{
  "eventType": "list",
  "ports": [
    {
      "address": "/dev/ttyACM0",
      "label": "/dev/ttyACM0",
      "hardwareId": "EBEABFD6514D32364E202020FF10181E",
      "properties": {
        "pid": "0x804e",
        "vid": "0x2341",
        "serialNumber": "EBEABFD6514D32364E202020FF10181E"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
  ]
}
```

This change may require an update to the [Pluggable Discovery](https://arduino.github.io/arduino-cli/0.19/pluggable-discovery-specification/) specification.

### Describe the current behavior

When a board changes the port after a reset, there is no reliable way to determine if the new port belongs to the same board.
With this proposal implemented the IDE may exploit the `hardwareId` field to reselect the correct port automatically.

### serial-discovery version

latest

### Operating system

N/A

### Operating system version

N/A

### Additional context

cc @pnndra @kittaakos @per1234 @umbynos 

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/serial-discovery/issues?q=)
- [X] I verified the feature was still missing when using the latest version
- [X] My request contains all necessary details

umbynos, 2023-01-24:
@cmaglie is the serial number exposed even in case of the network discovery? If it is not then we should start from the serial-discovery and then track the network-discovery part in another task.

umbynos, 2023-01-30:
TODO:

closed 2023-02-09 by cmaglie