arduino/arduino-cli #2039: Segmentation violation using a crafted sketch file and preprocess flag

luigigubello, 2023-01-19:
### Describe the problem

Arduino CLI returns a segmentation violation bug during the compilation if **preprocess** flag is `True` and the sketch filename contains brackets (e.g. `{filename}.ino`).
If you are using Arduino CLI as a gRPC server or service, this bug leads to the server crash (DoS).

### To reproduce

Tested on macOS.

**How to reproduce:**

- Enable arduino-cli gRPC server: `arduino-cli daemon --debug`
- Launch the bash script `arduino-cli-029-grpc-dos.sh` (attached).

**arduino-cli-029-grpc-dos.sh**

```
grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Create

grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 1}}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Init

grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 1}, "sketch_dir": "/tmp", "sketch_name": "{fooSketch}"}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.NewSketch

grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -proto cc/arduino/cli/commands/v1/compile.proto \
  -d '{"instance": {"id": 1}, "fqbn": "arduino:mbed_portenta:portenta_x8", "preprocess": true, "sketch_path": "/tmp/{fooSketch}/{fooSketch}.ino"}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Compile
  ```

### Expected behavior

The Arduino CLI should not crash, I guess.

### Arduino CLI version

0.29

### Operating system

macOS

### Operating system version

13

### Additional context

**Video PoC**

https://user-images.githubusercontent.com/27034386/213562035-cd85bfd6-dd30-4fdb-abab-9f6394cd063f.mp4

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2023-03-13:
Hi @luigigubello!

I can no longer reproduce the issue with the latest 0.31.0, in this release, a more strict file name validation is in place, `NewSketch` will not accept filenames with braces:
```
>>> NewSketch(instance:{id:1}  sketch_name:"{fooSketch}"  sketch_dir:"/tmp/cli2257049031/Arduino")
rpc error:
  code = Unknown
  desc = Can't create sketch: invalid sketch name "{fooSketch}": the first character must be alphanumeric, the following ones can also contain "_", "-", and ".".
```

BTW, even if we manually craft such a sketch I see that the Compile call will not crash anymore, with the latest build I get:
```
$ arduino-cli compile --preprocess "/tmp/{fooSketch}/{fooSketch}.ino" -b arduino:mbed_portenta:portenta_x8
arm-none-eabi-g++: error: /tmp/arduino/sketches/0353E36944FDE7020146CDA6A3C7BC09/sketch/.ino.cpp: No such file or directory
arm-none-eabi-g++: warning: '-x c++' after last input file has no effect
arm-none-eabi-g++: fatal error: no input files
compilation terminated.

Piattaforma utilizzata Versione Percorso                                                              
arduino:mbed_portenta  4.0.2    /home/megabug/.arduino15/packages/arduino/hardware/mbed_portenta/4.0.2
Si è verificato un errore durante la compilazione: exit status 1
$
```

instead using 0.29.0:

```
$ arduino-cli compile --preprocess "/tmp/{fooSketch}/{fooSketch}.ino" -b arduino:mbed_portenta:portenta_x8
arm-none-eabi-g++: error: /tmp/arduino-sketch-0353E36944FDE7020146CDA6A3C7BC09/sketch/.ino.cpp: No such file or directory
arm-none-eabi-g++: warning: '-x c++' after last input file has no effect
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0xbfd1e1]

goroutine 1 [running]:
github.com/arduino/arduino-cli/commands/compile.Compile({0x16ee340?, 0xe456ad?}, 0xc006112160, {0x10ebd80?, 0xc000014018}, {0x10ebd80?, 0xc000014020}, 0x0, 0xd1?)
	/home/megabug/Workspace/arduino-cli/commands/compile/compile.go:198 +0x1881
github.com/arduino/arduino-cli/cli/compile.runCompileCommand(0xc00037c000?, {0xc0002b6900, 0x1, 0x4?})
	/home/megabug/Workspace/arduino-cli/cli/compile/compile.go:235 +0xa9b
github.com/spf13/cobra.(*Command).execute(0xc00037c000, {0xc0002b68c0, 0x4, 0x4})
	/home/megabug/Software/go1.19.4/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:860 +0x663
github.com/spf13/cobra.(*Command).ExecuteC(0xc0002b2780)
	/home/megabug/Software/go1.19.4/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x3bd
github.com/spf13/cobra.(*Command).Execute(...)
	/home/megabug/Software/go1.19.4/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:902
main.main()
	/home/megabug/Workspace/arduino-cli/main.go:31 +0x77
```

the bug was in the error printing:

```go
compileErr := builder.RunPreprocess(builderCtx)
if compileErr != nil {
    compileErr = &arduino.CompileFailedError{Message: err.Error()}     <----- this must be compileErr, not err
}
```

This has been fixed in https://github.com/arduino/arduino-cli/pull/2035/files#diff-3f49374ea3ccca90f66635310e4bfb40a4aa48a800fdcba1ec704f51d25cd1e1R214

luigigubello, 2023-06-09:
I missed this comment, I will try next week :)

luigigubello, 2023-07-04:
LGTM :)