arduino/arduino-cli #2040: IDE recompiles whole project every time even if no changes were made

benapetr, 2023-01-22:
### Describe the problem

Right now IDE always recompiles whole project, every single file, no matter if changes were made or not. This makes large projects that compile tens of minutes extremely hard to develop as not only every minor change requires recompiling everything, but in fact even uploading to multiple devices requires recompiling each time.

### To reproduce

Just compile/verify anything that was already compiled

### Expected behavior

Don't recompile what didn't change. This is standard behaviour of most proper IDEs. Just compile changed files.

### Arduino IDE version

2.0.3

### Operating system

macOS

### Operating system version

10.14.6 (18G9323)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-22:
Hi @benapetr. Thanks for your report.

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Select **File > Preferences** from the Arduino IDE menus.
1. Check the box next to "**Show verbose output during: > ☐ compilation**"
1. Click the **<kbd>OK</kbd>** button.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to finish.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
   **ⓘ** This second compilation is done because we want to see the behavior when recompiling.
1. Wait for the compilation to finish.
1. Right click on the black "**Output**" panel at the bottom of the Arduino IDE window.
1. From the context menu, click "**Copy All**".
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

benapetr, 2023-01-22:
OK it seems I was wrong, according to output it's using previous object files, although it's still very slow... There are no timestamps, but it takes long time

```
Using board 'generic' from platform in folder: /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1
Using core 'esp8266' from platform in folder: /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1
/Users/petrbena/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/mkbuildoptglobals.py "/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build" 10607 /private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1 /private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/core/build.opt /Users/petrbena/Documents/Arduino/ESPuOS/ESPuOS.ino.globals.h /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266/CommonHFile.h
Note: optional global include file '/Users/petrbena/Documents/Arduino/ESPuOS/ESPuOS.ino.globals.h' does not exist.
  Read more at https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html
Detecting libraries used...
/Users/petrbena/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @/private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/core/build.opt -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/include -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/lwip2/include -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/libc/xtensa-lx106-elf/include -I/private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/core -c @/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/warnings/none-g++ -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" "-DARDUINO_BOARD_ID=\"generic\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266 -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/variants/generic /private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/sketch/ESPuOS.ino.cpp -o /dev/null
Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]
ResolveLibrary(ESP8266WiFi.h)
  -> candidates: [ESP8266WiFi@1.0]
/Users/petrbena/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @/private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/core/build.opt -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/include -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/lwip2/include -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/libc/xtensa-lx106-elf/include -I/private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/core -c @/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/warnings/none-g++ -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" "-DARDUINO_BOARD_ID=\"generic\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266 -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/variants/generic -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266WiFi/src /private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/sketch/ESPuOS.ino.cpp -o /dev/null
Alternatives for SoftwareSerial.h: [EspSoftwareSerial@7.0.0]
ResolveLibrary(SoftwareSerial.h)
  -> candidates: [EspSoftwareSerial@7.0.0]
/Users/petrbena/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @/private/var/folders/ns/f2w7g2351xn_v899jdwpp6f40000gn/T/arduino-sketch-E7E9AEFD3A84CE2AD665E95AC39F7FA1/core/build.opt -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/include -I/Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3. [... cut ...]
Using library EEPROM at version 1.0 in folder: /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/libraries/EEPROM 
Using library ESP8266HTTPClient at version 1.2 in folder: /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266HTTPClient 
Using library LittleFS at version 0.1.0 in folder: /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/libraries/LittleFS 
Using library ESP8266mDNS at version 1.2 in folder: /Users/petrbena/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS 

```

closed 2023-01-22 by benapetr

weswitt, 2023-01-23:
this is likely an issue with arduinocli. i opened an issue for this a while ago: https://github.com/arduino/arduino-cli/issues/1996.