arduino/arduino-cli #2043: The `sketch new` command ignores the sketch folder name specification

kittaakos, 2023-01-23:
### Describe the problem

I can use the CLI `sketch new` to create invalid sketches which will not compile. The little sister task of this one is arduino/arduino-ide#1599.

### To reproduce

```
./arduino-cli version                                                          
arduino-cli  Version: git-snapshot Commit: a58d5adb Date: 2023-01-23T09:37:23Z
```

```
./arduino-cli sketch new "/Users/a.kitta/Documents/Arduino/\`~%^&()=+[{]};',.cpp"
Sketch created in: /Users/a.kitta/Documents/Arduino/`~%^&()=+[{]};',.cpp
```

```
cat ~/Documents/Arduino/\`\~%\^\&\(\)=+\[\{\]\}\;\',.cpp/\`\~%\^\&\(\)=+\[\{\]\}\;\',.cpp.ino

void setup() {
}

void loop() {
}
```

```
./arduino-cli compile -b arduino:avr:uno  "/Users/a.kitta/Documents/Arduino/\`~%^&()=+[{]};',.cpp"
avr-g++: error: /private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino/sketch-94408ACC1CA44D8729BBBE1057EBD2E8/sketch/`~%^&()=+[;',.cpp.ino.cpp: No such file or directory
avr-g++: warning: '-x c++' after last input file has no effect
avr-g++: fatal error: no input files
compilation terminated.
```

Used platform Version Path                                                                
arduino:avr   1.8.6   /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/avr/1.8.6
Error during build: exit status 1

```

### Expected behavior

`new sketch` fails when the sketch's name is invalid. https://arduino.github.io/arduino-cli/dev/sketch-specification/#sketch-root-folder

### Arduino CLI version

a58d5adb

### Operating system

macOS

### Operating system version

12.5.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

umbynos, 2023-01-23:
This should be a matter of implementing a regex to check if the sketch name is a valid one

umbynos, 2023-01-24:
AC: if a name that does not respect the regex is used an appropriate error is returned. Otherwise the sketch name is accepted.

closed 2023-01-30 by Bikappa