arduino/arduino-cli #2056: First lines of Serial output are flushed

ubidefeo, 2023-01-31:
### Describe the problem

This is mostly noticeable from the IDE 2.0, because users of `arduino-cli monitor` such as myself tend to consider that some of the output might get lost between issuing the `upload` command and launching the `monitor`.

After the `Serial.begin(BAUD_RATE)` command, anything printed gets lost until the Serial starts receiving new data.

### To reproduce

```c++
unsigned int counter = 0;
void setup(){
    Serial.begin(9600);
    Serial.println("hello");
    delay(3000);
    Serial.println("end of setup");
}

void loop(){
    counter++;
    if(counter < 10){
      Serial.println(counter);
    }
    
}
```

While @umbynos, @facchinm and I were looking at the code where the port `touch` at 1200b happens,
we realised that the order of operations might need to be reworked in order for the buffer not to be forcefully flushed.

That was a while back, so I can't exactly recall what we thought we'd do to work this out,
but hope this example for reproducibility helps track it down

### Expected behavior

Everything printed after the serial port is open should be available to the serial monitor

### Arduino CLI version

0.29.0

### Operating system

macOS

### Operating system version

12.6.2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

umbynos, 2023-02-06:
@ubidefeo talk with @facchinm

cmaglie, 2023-02-15:
Fixed by arduino/serial-monitor#39

closed 2023-02-15 by cmaglie