arduino/arduino-cli #2098: Board discovery lists Intel Active Management Technology - SOL

bradmartin333, 2023-03-08:
### Describe the problem

On some company PCs with no boards connected, `board list` discovers a COM port of Protocol `serial`. The COM port is assigned to Intel Active Management Technology - SOL and does not show up in the `usbdeview` applicaton.

### To reproduce

- Unplug all USB devices
- Open device manager
- Expand Ports (COM & LPT) if available
- Determine if the specific PC has `Intel Active Management Technology - SOL (COM3)`
- If so, run `.\arduino-cli.exe board list` in PowerShell
- Observe detection of COM3 in CLI output as `COM3   serial   Serial Port   Unknown`

### Expected behavior

CLI would output `No boards found.`

### Arduino CLI version

nightly-20230308

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

This occurs on a coworkers computer. When the board is plugged in, it mounts on COM3 as well, which is confusing. At that point, the CLI output says `serial (USB)` instead of `serial`. I am going to fix it on my end by searching for `(USB)` in the output string and failing if it is not present in the string.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

bradmartin333, 2023-03-09:
This is how I am currently getting the first valid Arduino port with PowerShell

``` PowerShell
# Isolate the first Arduino port that shows up
Write-Host 'Looking for Arduinos'
$port = 'NONE'
$list = .\arduino-cli.exe board list
for (($i = 1); $i -lt $list.Length; $i++)
{
    if ($list[$i] -is [char]) {
        break # "No boards found." string
    }
    if ($list[$i].Contains('(USB)')) {
        # Use regex to get the port
        $matches = $list[$i] | Select-String -Pattern 'COM\d+'
        $port = $matches.Matches[0].Value
        Write-Host 'Found Arduino on' $port
    }
}
if ($port -eq 'NONE') {
    Write-Host 'No Arduinos found... exiting'
    Start-Sleep 3
    exit
}
```

cmaglie, 2023-06-09:
I never heard about this "Intel Active Management Technology - SOL" device, after a brief search on Google I see that many people are getting this weird "COM3":

https://www.dslreports.com/forum/r27379036-WIN7-Intel-Active-Management-Technology-SOL-COM3
https://social.technet.microsoft.com/Forums/windows/en-US/f6e23a61-92d8-4e21-ba30-b65ec3e29542/intel-active-management-technology-sol-com3?forum=w7itprohardware

Anyway, the device COM3 is present, and the CLI correctly lists it... I don't think it's a CLI issue, did you agree?

bradmartin333, 2023-06-09:
Ya, that's fine. No issues with the workaround either.

closed 2023-06-09 by bradmartin333