arduino/arduino-cli #2104: Invalid JSON output for outdated command

Renaud11232, 2023-03-10:
### Describe the problem

When running the `outdated` command with `--format json`, the produced output is not valid JSON. It looks like it contains two arrays.

### To reproduce

Install an outdated library ie: 
```bash
arduino-cli lib install "FastLED@3.3.2"
```
Then run : 
```bash
arduino-cli --format json outdated
```
Which produces the following output : 
```json
[]
[
  {
    "library": {
      "name": "FastLED",
      "author": "Daniel Garcia",
      "maintainer": "Daniel Garcia \u003cdgarcia@fastled.io\u003e",
      "sentence": "Multi-platform library for controlling dozens of different types of LEDs along with optimized math, effect, and noise functions.",
      "paragraph": "Multi-platform library for controlling dozens of different types of LEDs along with optimized math, effect, and noise functions.",
      "website": "https://github.com/FastLED/FastLED",
      "category": "Display",
      "architectures": [
        "*"
      ],
      "install_dir": "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED",
      "source_dir": "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED",
      "version": "3.3.2",
      "license": "Unspecified",
      "location": 1,
      "examples": [
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\AnalogOutput",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Blink",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\ColorPalette",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\ColorTemperature",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Cylon",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\DemoReel100",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Fire2012",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Fire2012WithPalette",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\FirstLight",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Multiple\\ArrayOfLedArrays",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Multiple\\MirroringSample",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Multiple\\MultiArrays",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Multiple\\MultipleStripsInOneArray",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Multiple\\OctoWS2811Demo",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Multiple\\ParallelOutputDemo",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Noise",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\NoisePlayground",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\NoisePlusPalette",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Pintest",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\Ports\\PJRCSpectrumAnalyzer",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\RGBCalibrate",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\RGBSetDemo",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\SmartMatrix",
        "C:\\Users\\gaspa\\Documents\\Arduino\\libraries\\FastLED\\examples\\XYMatrix"
      ],
      "provides_includes": [
        "FastLED.h",
        "bitswap.h",
        "chipsets.h",
        "color.h",
        "colorpalettes.h",
        "colorutils.h",
        "controller.h",
        "cpp_compat.h",
        "dmx.h",
        "fastled_config.h",
        "fastled_delay.h",
        "fastled_progmem.h",
        "fastpin.h",
        "fastspi.h",
        "fastspi_bitbang.h",
        "fastspi_dma.h",
        "fastspi_nop.h",
        "fastspi_ref.h",
        "fastspi_types.h",
        "hsv2rgb.h",
        "led_sysdefs.h",
        "lib8tion.h",
        "noise.h",
        "pixelset.h",
        "pixeltypes.h",
        "platforms.h",
        "power_mgt.h"
      ]
    },
    "release": {
      "author": "Daniel Garcia",
      "version": "3.5.0",
      "maintainer": "Daniel Garcia \u003cdgarcia@fastled.io\u003e",
      "sentence": "Multi-platform library for controlling dozens of different types of LEDs along with optimized math, effect, and noise functions.",
      "paragraph": "Multi-platform library for controlling dozens of different types of LEDs along with optimized math, effect, and noise functions.",
      "website": "https://github.com/FastLED/FastLED",
      "category": "Display",
      "architectures": [
        "*"
      ],
      "types": [
        "Contributed"
      ]
    }
  }
]
```

### Expected behavior

The output should be valid JSON.

### Arduino CLI version

arduino-cli.exe  Version: nightly-20230310 Commit: fed439a Date: 2023-03-10T01:35:49Z

### Operating system

Windows

### Operating system version

Windows 11 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

lluiscampos, 2023-03-14:
I looked into this (I was just curious) and the issue is that since the simplification of `outdated` command (commit [here](https://github.com/arduino/arduino-cli/commit/61ba6854c#diff-b6f5262ac27b9c48547e29ba62c67496cb86476f3755d5e272132fb1d50f0fcc), by @cmaglie), what the code is doing is first to list the core packages that are upgradable (same as `arduino-cli core list --updatable`) and then libraries that are upgradable (`./arduino-cli lib list --updatable`).

And while this works nicely for the text format, it does not produce valid JSON, because it just prints two separate JSON objects (in your example, an empty `[]` for core packages and then the libraries part).

The fix that I can think of is refactor a bit the code so that `internal/cli/outdated` can get the slices returned from both `internal/cli/core` and `internal/cli/lib` packages, have its own implementation of `Result interface` and call `feedback.PrintResult` directly with the appended slices.

@cmaglie, can you please comment on my suggestion? Thanks!

cmaglie, 2023-03-15:
> The fix that I can think of is refactor a bit the code so that internal/cli/outdated can get the slices returned from both internal/cli/core and internal/cli/lib packages, have its own implementation of Result interface and call feedback.PrintResult directly with the appended slices.

That's correct, but I don't know how much refactoring is required to do so. I see that you made a PR already, I guess you tried implementing the idea above, I'll take a look at it ASAP.

lluiscampos, 2023-03-16:
@cmaglie Thanks! Yes, I got started before your feedback as it seemed quite right. The PR is ready for review now :mag:

PR #2111 (2023-03-17): Fix invalid JSON and YAML outputs for `outdated` command
Fixes issue #2104

The code for `internal/cli/core` and `internal/cli/lib` has been
refactored so that we can use only the `Get` functions from
`internal/cli/outdated` package and compose there composited object.

For regular text output, the new table will have some extra fields that
either for platforms or for libraries will be blank.

For JSON and YAML output, the resulting object will have the top-level
keys `platforms` and `libraries` which contain, respectively, the array
of outaded platforms and outdated libraries.

## Please chec [... cut]
closed 2023-03-17 by cmaglie