arduino/arduino-cli #2143: brew install of arduino-cli leads to `discovery builtin:mdns-discovery not found` error

aliphys, 2023-04-11:
### Describe the problem

Upon installing the arduino-cli via [brew](https://arduino.github.io/arduino-cli/0.32/installation/#install-via-homebrew-macoslinux), an `Error initializing instance: Error loading hardware platform: discovery builtin:mdns-discovery not found` error is displayed when the `arduino-cli core list` command is entered into the terminal.

[image]

### To reproduce

1. Install `brew` and `gcc`from the terminal
```
sudo apt install curl

/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"~

(echo; echo 'eval "$(/home/linuxbrew/.linuxbrew/bin/brew shellenv)"') >> /home/ali/.profile
eval "$(/home/linuxbrew/.linuxbrew/bin/brew shellenv)"

sudo apt-get install build-essential
brew install gcc
```
2. Install `arduino-cli`
```
brew update
brew install arduino-cli

```
3.  Run command `arduino-cli core list`
```
$ arduino-cli core list
Error initializing instance: Error loading hardware platform: discovery builtin:mdns-discovery not found
Error initializing instance: Error loading hardware platform: discovery builtin:mdns-discovery not found
Error initializing instance: Error loading hardware platform: discovery builtin:mdns-discovery not found
Error initializing instance: Error loading hardware platform: discovery builtin:mdns-discovery not found
ID                Installed Latest Name                                        
arduino:avr       1.8.3     1.8.3  Arduino AVR Boards                          
arduino:mbed_nano 2.0.0     2.0.0  Arduino Mbed OS Nano Boards                 
arduino:samd      1.8.11    1.8.11 Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```

### Expected behavior

```
$ arduino-cli core list
ID                Installed Latest Name                                        
arduino:avr       1.8.3     1.8.6  Arduino AVR Boards                          
arduino:mbed_nano 2.0.0     4.0.2  Arduino Mbed OS Nano Boards                 
arduino:samd      1.8.11    1.8.13 Arduino SAMD Boards (32-bits ARM Cortex-M0+)

```

### Arduino CLI version

arduino-cli  Version: 0.32.1 Commit: a8787b82 Date: 2023-04-07T13:55:06Z

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Additional context

It is possible to solve this by running the `arduino-cli core update-index` command as mentioned by a Forum user [here](https://forum.arduino.cc/t/getting-discovery-not-found-trying-to-upload-with-arduino-cli/950526/5). Then, the next time `arduino-cli core list` is run, the required tools are installed. 
 
[image]

Ideally this should not be required and this step should happen without user intervention. 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

sminogue, 2023-07-02:
This also happened to me when running `arduino-cli board list` after installation. The "workaround" above solved the problem.

I would recommend adding the command `arduino-cli core update-index` to the brew installation.

umbynos, 2023-08-23:
@aliphys does this happens also with the latest version of the cli?

aliphys, 2023-08-23:
I've installed the IDE several times since I made this bug report. Haven't had this issue reoccur.

I guess we can close this issue. 🙂