arduino/arduino-cli #2154: Arduino CLI (2.x) missing --get-pref (?) & –upload behavour changed (?)

James-4DSystems, 2023-04-20:
### Describe the problem

Firstly, We use arduino_debug (Arduino 1.x CLI) --get-pref to find out a few pieces of information about the current board settings, eg
'target_package='
'target_platform='
'custom_cpu='
'board='

How can we do this with with the 2.x Arduino CLI, or is it simply missing and not possible?

It would appear to live under Arduino-CLI config dump but we couldn't see it there

--

Secondly, The Arduino 1.x CLI (Arduino_debug) does a 'recompile if source changed' when using –upload
The Arduino 2.0.4 CLI doesn't (equivalent command is just upload, so not that different)

Is this working 'as designed' or is it a bug?

### To reproduce

Hopefully obvious as per the description

### Expected behavior

We would have expected the 2.x CLI to do the same or equivalent as the 1.x CLI (arduino_debug)
If there are name changes etc, can you please advise what they are as we cannot find them

### Arduino CLI version

arduino-cli  Version: nightly-20230419 Commit: bfb5f3f Date: 2023-04-19T01:33:22Z

### Operating system

Windows

### Operating system version

10 & 11

### Additional context

We use the CLI to compile via our own IDE, and we are currently trying to get the IDE to compile via the 2.x CLI as many customers have moved over to 2.x from 1.x, and we are trying to support it but finding things are just not working.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

James-4DSystems, 2023-04-21:
Thanks for the reply

> Back to your question, it's not possible to know the board selection from the Arduino IDE 2.0 using arduino-cli, why do you need this info?

Because we use this as the default starting point for our IDE, no biggie

> Secondly, The Arduino 1.x CLI (Arduino_debug) does a 'recompile if source changed' when using –upload
> The Arduino 2.0.4 CLI doesn't (equivalent command is just upload, so not that different)
> Also, the 2.0.4 CLI simply complains "Compiled sketch not found in C:\Users\xxxx\AppData\Local\Temp\arduino..." if the sketch has not been compiled in the current session
> Is this working 'as designed' or is it a bug?

Any answer to this part?

cmaglie, 2023-04-21:
> Secondly, The Arduino 1.x CLI (Arduino_debug) does a 'recompile if source changed' when using –upload
The Arduino 2.0.4 CLI doesn't (equivalent command is just upload, so not that different)
Also, the 2.0.4 CLI simply complains "Compiled sketch not found in C:\Users\xxxx\AppData\Local\Temp\arduino..." if the sketch has not been compiled in the current session
Is this working 'as designed' or is it a bug?

The error happens because you must first compile the sketch before uploading. You can do a combined compile+upload with the following syntax (change FQBN and PORT depending on your board):

```
arduino-cli compile --fqbn arduino:samd:mkr1000 -u -p /dev/ttyACM0 /path/to/the/sketch/
```

James-4DSystems, 2023-04-21:
> The error happens because you must first compile the sketch before uploading. You can do a combined compile+upload with the following syntax (change FQBN and PORT depending on your board):
> 
> arduino-cli compile --fqbn arduino:samd:mkr1000 -u -p /dev/ttyACM0 /path/to/the/sketch/

OK, so that functionality is different to arduino_Debug. can you confirm that that is by design?

Is there a reason the CLI has changed functionality like this from arduino_Debug, rather than making it backwards compatible? Having to support both now is a bit of a pain.

Thanks

cmaglie, 2023-04-21:
> OK, so that functionality is different to arduino_Debug. can you confirm that that is by design?

Yes, it is.

> Is there a reason the CLI has changed functionality like this from arduino_Debug, rather than making it backwards compatible? Having to support both now is a bit of a pain.

There are many reasons to make it this way, any way, there is no point in discussing it here, as I said:
* Arduino IDE 2.0.0 is a major 2.x.x release, basically rewritten from scratch with modern frameworks... breaking changes are to be expected.
* Arduino CLI is not the Arduino IDE (even if Arduino IDE 2.x.x bundles the Arduino CLI to work)

closed 2023-04-23 by James-4DSystems