arduino/arduino-cli #2158: connection refused when updating index

Sw3d15h-F1s4, 2023-04-24:
### Describe the problem

When running `arduino-cli core update-index`

`Downloading index: library_index.tar.bz2 Get "https://downloads.arduino.cc/libraries/library_index.tar.bz2": dial tcp: lookup downloads.arduino.cc on [::1]:53: read udp [::1]:54136->[::1]:53: read: connection refused
Error updating indexes: Error downloading index 'https://downloads.arduino.cc/libraries/library_index.tar.bz2': Get "https://downloads.arduino.cc/libraries/library_index.tar.bz2": dial tcp: lookup downloads.arduino.cc on [::1]:53: read udp [::1]:54136->[::1]:53: read: connection refused`

### To reproduce

installed linux arm64 binary from arduino.github.io's installation instructions, moved it to /bin and executed. seems to work fine otherwise. ran arduino-cli config init as well.

### Expected behavior

i expect it to update the board index

### Arduino CLI version

arduino-cli  Version: 0.32.2 Commit: 2661f5d9 Date: 2023-04-12T09:41:53Z

### Operating system

Linux

### Operating system version

linux aarch64 5.4.147-qgki-25884040-abT733XXU2CWC5

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2023-04-28:
This looks like a network configuration problem:

```
dial tcp: lookup downloads.arduino.cc on [::1]:53: read udp [::1]:54136->[::1]:53: read: connection refused
```

Arduino CLI cannot resolve the name `downloads.arduino.cc`.
Does `ping downloads.arduino.cc` works?

codeandfire, 2023-05-11:
Hi, I'm having the same issue; tried the Nightly version, v0.32.2, and v0.31.0.

`ping downloads.arduino.cc` works, though it seems to be a bit slow:
```
PING downloads.arduino.cc.cdn.cloudflare.net (104.18.12.241) 56(84) bytes of data.
64 bytes from 104.18.12.241 (104.18.12.241): icmp_seq=1 ttl=53 time=38.7 ms
64 bytes from 104.18.12.241 (104.18.12.241): icmp_seq=2 ttl=53 time=41.3 ms
64 bytes from 104.18.12.241 (104.18.12.241): icmp_seq=3 ttl=53 time=39.7 ms
^C64 bytes from 104.18.12.241: icmp_seq=4 ttl=53 time=37.8 ms

--- downloads.arduino.cc.cdn.cloudflare.net ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 23270ms
rtt min/avg/max/mdev = 37.827/39.388/41.293/1.288 ms
```

I'm also able to manually download that file (`https://downloads.arduino.cc/libraries/library_index.tar.bz2`), if that helps.

cmaglie, 2023-05-11:
Do you have an error similar to the following?

```
dial tcp: lookup downloads.arduino.cc on [::1]:53: read udp [::1]:54136->[::1]:53: read: connection refused
```

if yes, could you paste the output of:

```
sudo netstat -puntl
```

and

```
dig downloads.arduino.cc
```

codeandfire, 2023-05-11:
@cmaglie thanks for responding.

> Do you have an error similar to the following?
> 
> ```
> dial tcp: lookup downloads.arduino.cc on [::1]:53: read udp [::1]:54136->[::1]:53: read: connection refused
> ```

Yes, precisely.

> if yes, could you paste the output of:
> 
> ```
> sudo netstat -puntl
> ```

```
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.54:53           0.0.0.0:*               LISTEN      428/systemd-resolve 
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      428/systemd-resolve 
tcp        0      0 0.0.0.0:5355            0.0.0.0:*               LISTEN      428/systemd-resolve 
tcp6       0      0 :::5355                 :::*                    LISTEN      428/systemd-resolve 
udp        0      0 127.0.0.54:53           0.0.0.0:*                           428/systemd-resolve 
udp        0      0 127.0.0.53:53           0.0.0.0:*                           428/systemd-resolve 
udp        0      0 192.168.1.7:68          0.0.0.0:*                           488/iwd             
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           428/systemd-resolve 
udp        0      0 0.0.0.0:5355            0.0.0.0:*                           428/systemd-resolve 
udp6       0      0 :::5353                 :::*                                428/systemd-resolve 
udp6       0      0 :::5355                 :::*                                428/systemd-resolve 
```

> and
> 
> ```
> dig downloads.arduino.cc
> ```

```
;; communications error to ::1#53: connection refused
;; communications error to ::1#53: connection refused
;; communications error to ::1#53: connection refused
;; communications error to 127.0.0.1#53: connection refused

; <<>> DiG 9.18.14 <<>> downloads.arduino.cc
;; global options: +cmd
;; no servers could be reached
```

cmaglie, 2023-05-12:
Also knowing the Linux distribution may help to find if there are open bugs.

codeandfire, 2023-05-12:
Thank you for the explanation! I have a fresh Arch Linux installation so `/etc/resolv.conf` is basically empty:

```
$ cat /etc/resolv.conf 
# Resolver configuration file.
# See resolv.conf(5) for details.
```

Is that the problem? I am not facing network issues anywhere else - am able to browse normally, etc. - so I wasn't suspecting network configuration issues on my end.

Sw3d15h-F1s4, 2023-05-15:
Ok, so after some searching, reading, and learning, I think I know what my issue is. I'm trying to run the arduino cli on Android through Termux, which has many differences from regular Linux (such as no etc/resolv.conf and using different c libraries instead of gcc). Now I don't need to compile, I was just trying to use arduino-cli to set up arduino-language-server for neovim (surprisingly neovim is quite nice to use on Android, other than lsp installation and setup being harder than usual) since I wanted to be able to contribute to my team's arduino sketches while on the move at college. Yeah, ik, dumb to run this on android, whatever. For anyone in the same boat, platformio works a little better for Android since it uses regular .cpp files, and you can set up clangd or whatever else like normal (only get a mild issue of no thread api but again you shouldn't need to compile from an android device, like how are you going to push code to the board your USB ports are a challenge to access through termux). Anyway, sorry for the trouble. I'm a noob to Linux, and figured throwing myself in the deep end on Termux might learn me something. Hope yall have a nice day.

codeandfire, 2023-05-16:
@cmaglie symlinking `/etc/resolv.conf` to `/run/systemd/resolve/stub-resolv.conf` worked for me ... Thank you very much for your help!

cmaglie, 2023-05-16:
Thanks for the feedback!
Since it seems resolved in one way or another, I'm closing the issue.

closed 2023-05-16 by cmaglie