arduino/arduino-cli #2180: Compiler uses wrong library path when compiling with a local profile

TimH1502, 2023-05-19:
### Describe the problem

When i use the arduino-cli 0.32 (and nightly-20230519 Commit: 62f5caf Date: 2023-05-19T01:33:35Z) the compiler uses a wrong local library path (partly right) and the following error is shown

Error during build: 
CreateFile toolchain\internal\ArduinoModbus_1.0.8_4504f0c6d4bba471\ArduinoModbus\src\toolchain\internal\ArduinoModbus_1.0.8_4504f0c6d4bba471\ArduinoModbus\src\ModbusClient.cpp: The system cannot find the path specified.

The arduino-cli is able to download the mbed-core and the libraries described in the "sketch.yaml".

The command --library or --libraries does not work with --profile

### To reproduce

- use the project (zip file): [MyFirstSketch.zip](https://github.com/arduino/arduino-cli/files/11517452/MyFirstSketch.zip)
- use the arduino-cli 0.32 (and nightly-20230519 Commit: 62f5caf date: 2023-05-19T01:33:35Z)
- arduino-cli.exe compile -b arduino:mbed_portenta:envie_m7 --profile M7Core -v --export-binaries --config-file  .\arduino-cli.yaml
[MyFirstSketch.zip](https://github.com/arduino/arduino-cli/files/11517452/MyFirstSketch.zip)

### Expected behavior

Path which should be used: 

toolchain\internal\ArduinoModbus_1.0.8_4504f0c6d4bba471\ArduinoModbus\src\ModbusClient.cpp

### Arduino CLI version

nightly-20230519 Commit: 62f5caf date: 2023-05-19T01:33:35Z

### Operating system

Windows

### Operating system version

Windows 11 22621.1702 (22H2) and Windows 10 19045.2846 (22H2)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2023-08-16:
I've created a patch here #2265.

Anyway I see that  inside your sketch you have a local config file that set the data directory and the user directory as a sub-directory of the sketch itself:

```
directories:
  data: toolchain
  user: toolchain
metrics:
  enabled: false
  always_export_binaries: true
```

thic config will make the arduino-cli download again all the tools inside the sketch, wasting a lot of disk space.
This is useless since the arduino-cli already uses a private space to download the platforms/tools needed for a profile, and this space is shared among all sketches, and it won't interfere with the platforms/tools installed globally (for use without profiles).

So my recommendation is to simply remove the local `arduino-cli.yamll` config and the `toolchain` folder.

closed 2023-08-17 by cmaglie

TimH1502, 2023-08-18:
First off all thank you for your patch.
Your concern about the disk space is justified.

If use a different folder structure: `C:\Users\USERNAME\Downloads\Projects\MyFirstProject\MyFirstProject.ino`
the compiler can't access the `<bits/c++config.h>`

Complete error message: 

> In file included from c:\users\USERNAME\downloads\projects\myfirstsketch\toolchain\internal\arduino_arm-none-eabi-gcc_7-2017q4_7b7be9f526b2cb64\arm-none-eabi\include\c++\7.2.1\math.h:36:0, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_def.h:33, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_rcc.h:29, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/stm32h7xx_hal_conf.h:398, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal.h:30, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h7xx.h:223, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/cmsis.h:22, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/mbed/targets/TARGET_STM/PinNamesTypes.h:33, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\variants\PORTENTA_H7_M7/pinmode_arduino.h:24, 
                 from toolchain\internal\arduino_mbed_portenta_4.0.4_038db0ce27dff130\cores\arduino/Arduino.h:26, 
                 from C:\Users\USERNAME\AppData\Local\Temp\arduino\sketches\362617C8BF6AEA5AA483D0348C335006\sketch\MyFirstSketch.ino.cpp:1: 
c:\users\USERNAME\downloads\projects\myfirstsketch\toolchain\internal\arduino_arm-none-eabi-gcc_7-2017q4_7b7be9f526b2cb64\arm-none-eabi\include\c++\7.2.1\cmath:41:10: fatal error: bits/c++config.h: No such file or directory 
#include <bits/c++config.h> 
          ^~~~~~~~~~~~~~~~~~ 
compilation terminated.

Using the following structure: `C:\Users\USERNAME\Downloads\MyFirstProject\MyFirstProject.ino`
the compilation succeeds.

cmaglie, 2023-08-21:
Unfortunately, this is a known issue of `arm-none-eabi-gcc 7-2017q4`, which cannot handle paths with more than ~150 chars on Windows.

AFAIK the last time I checked (about a year ago) the issue was still present upstream, I don't know if it has been fixed in the meantime in the most recent version of arm-gcc.

In any case, this is not related to the CLI but to the compiler.

TimH1502, 2023-08-25:
Okay if I move the project it seems to work as expected.
Are there any plans in the future to change the gcc version to 2021?