arduino/arduino-cli #2197: CLI provides insufficient BoardDetails#build_properties info when the platform is installed via Git

dankeboy36, 2023-06-01:
### Describe the problem

I have noticed when a platform is installed from git, instead of the _Boards Manager_ (`./arduino-cli core install id`), then the `BoardDetails#build_properties` fails to provide sufficient information to calculate required tool paths. See https://github.com/arduino/arduino-ide/pull/2071#issuecomment-1560534977.

### To reproduce

Install the esp32 [platform manually via Git](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation), and run the get the board details for `esp32:esp32:esp32` (the FQBN can be `espressif:esp32:esp32` if you strictly follow the steps).

The missing `runtime.tools.*` is the real problem for the Git-installed platform:

```diff
341,381c340,355
<     "runtime.platform.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/hardware/esp32/2.0.9",
<     "runtime.hardware.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/hardware/esp32",
<     "build.core.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/hardware/esp32/2.0.9/cores/esp32",
<     "build.system.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/hardware/esp32/2.0.9/system",
<     "build.variant.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/hardware/esp32/2.0.9/variants/esp32",
<     "runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp8266/tools/mkspiffs/3.1.0-gcc10.3-e5f9fec",
<     "runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp8266/tools/mklittlefs/3.1.0-gcc10.3-e5f9fec",
<     "runtime.tools.esptool_py.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/esptool_py/4.5.1",
<     "runtime.tools.esptool_py-4.5.1.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/esptool_py/4.5.1",
<     "runtime.tools.mklittlefs.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933",
<     "runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933",
<     "runtime.tools.mkspiffs.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/mkspiffs/0.2.3",
<     "runtime.tools.mkspiffs-0.2.3.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/mkspiffs/0.2.3",
<     "runtime.tools.openocd-esp32.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026",
<     "runtime.tools.openocd-esp32-v0.11.0-esp32-20221026.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026",
<     "runtime.tools.riscv32-esp-elf-gcc.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.riscv32-esp-elf-gdb.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/11.2_20220823",
<     "runtime.tools.riscv32-esp-elf-gdb-11.2_20220823.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/11.2_20220823",
<     "runtime.tools.xtensa-esp-elf-gdb.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823",
<     "runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823",
<     "runtime.tools.xtensa-esp32-elf-gcc.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.xtensa-esp32s2-elf-gcc.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.xtensa-esp32s3-elf-gcc.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0",
<     "runtime.tools.serial-monitor.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/serial-monitor/0.13.0",
<     "runtime.tools.serial-monitor-0.13.0.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/serial-monitor/0.13.0",
<     "runtime.tools.ctags.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/ctags/5.8-arduino11",
<     "runtime.tools.ctags-5.8-arduino11.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/ctags/5.8-arduino11",
<     "runtime.tools.mdns-discovery.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/mdns-discovery/1.0.9",
<     "runtime.tools.mdns-discovery-1.0.9.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/mdns-discovery/1.0.9",
<     "runtime.tools.xtensa-lx106-elf-gcc.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec",
<     "runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec",
<     "runtime.tools.serial-discovery.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/serial-discovery/1.4.0",
<     "runtime.tools.serial-discovery-1.4.0.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/builtin/tools/serial-discovery/1.4.0",
<     "runtime.tools.python3.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp8266/tools/python3/3.7.2-post1",
[... cut ...]
The `BoardDetails#build_proeprties` info is the same when the platform is installed via boards manager and Git. (Expanded path values could be different)

### Arduino CLI version

0.33.0

### Operating system

macOS

### Operating system version

11.7.7

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-01:
@dankeboy36 is this about the missing `tools.xtensa-esp-elf-gdb.path` and `tools.riscv32-esp-elf-gdb.path` properties? If so, this is because for some reason the `master` branch of the `espressif/arduino-esp32` repo is significantly outdated. It is missing this commit that is in the 2.0.9 release version you get from Boards Manager:

https://github.com/espressif/arduino-esp32/commit/8fca36592e0b5f429e5d2392a412e81422a1e8bf

Check out the `release/v2.x` branch in your `espressif/arduino-esp32` clone and then run the `tools/get.py` script again to update the toolchain. After you do that the `tools.xtensa-esp-elf-gdb.path` and `tools.riscv32-esp-elf-gdb.path` properties will be defined when using the manually installed ESP32 platform.

me-no-dev, 2023-06-01:
`runtime.tools.*` is defined by Arduino IDE and because it was causing clashes between core installed through the board manager and through git, we have recently removed those definitions in the git variant of `platform.txt`. What you actually need is `tools.*` which in both cases (git/board manager) will show the correct path to tools.

dankeboy36, 2023-06-02:
Thank you!

> is this about the missing `tools.xtensa-esp-elf-gdb.path` and `tools.riscv32-esp-elf-gdb.path` properties?

It's about all missing `runtime.tools.*` properties.

> Check out the `release/v2.x`

My tests check out the [`2.0.9` branch](https://github.com/espressif/arduino-esp32/tree/2.0.9) from [`espressif/arduino-esp32`](https://github.com/espressif/arduino-esp32) from the see the CI logs [here](https://github.com/dankeboy36/arduino-ide-esp-exception-decoder/actions/runs/5148558024/jobs/9270405117#step:8:222). It is [3670e2b](https://github.com/espressif/arduino-esp32/commit/3670e2bf2aca822f2e1225fdb0e0796e490005a8).

> What you actually need is `tools.*`

That is a super helpful remark. I made some progress, but somewhere, there is a bug. I will reuse snippets from the attached files to https://github.com/arduino/arduino-cli/issues/2197#issue-1736955123 to describe the situation. The FQBN is `esp32:esp32:esp32`. (I used the `esp32` folder name instead of `espressif` described in the [manual installation](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation). `esp32:esp32:esp32` is OK.)

For `esp32:esp32:esp32`, the tool lookup logic should look for `tools.xtensa-esp-elf-gdb.path` and
`tools.xtensa-esp32-elf-gcc.path`, and the DGB executable name is `xtensa-esp32-elf-gdb`. (Let's ignore the `runtime.tools.*` case for now.)

Relevant `build_properties` content when installed via CLI:

```json
"tools.xtensa-esp-elf-gdb.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823",
"tools.xtensa-esp32-elf-gcc.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/cli/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0",
```

The expected path is `cli/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823/bin/xtensa-esp32-elf-gdb`.

[image]

Relevant `build_properties` content when installed via Git:

```json
"tools.xtensa-esp-elf-gdb.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/git/Arduino/hardware/esp32/esp32/tools/xtensa-esp-elf-gdb",
"tools.xtensa-esp32-elf-gcc.path": "/Users/dankeboy36/Desktop/dev/arduino-ide-esp-exception-decoder/test-resources/envs/git/Arduino/hardware/esp32/esp32/tools/xtensa-esp32-elf",
```

When installed from Git, the CLI's board properties response expands the `tools.xtensa-esp-elf-gdb.path` to `./Arduino/hardware/esp32/esp32/tools/xtensa-esp-elf-gdb`, but this folder does not exist. It should be `./Arduino/hardware/esp32/xtensa-esp-elf-gdb`.

[image]

I do not know the details of the property expansion, but I think this is either a bug in the CLI or my tool lookup logic should handle corner cases which I wanted to avoid a rely solely on the CLI.

dankeboy36, 2023-06-02:
> Then you run `Arduino/hardware/espressif/esp32/tools/get.py`

This was the problem. When executing the `python` script, the `cwd` must be `Arduino/hardware/espressif/esp32/tools/`. I incorrectly set it to `Arduino/hardware/espressif/` in my tests. Everything is working great now.

Thank you!

It's not a CLI bug.

closed 2023-06-02 by dankeboy36