arduino/arduino-cli #2200: Use `"additionalProperties": false` in the configuration JSONSchema

cmaglie, 2023-06-05:
### Describe the request

from @kittaakos:

> I have noticed that the `additionalProperties` is absent from the schema. Hence, when generating types downstream, arbitrary properties are valid. Are they valid? In other words, is the "additionalProperties": false missing on purpose?

### Describe the current behavior

If an extraneous property is present in the config file of the CLI it is not detected as "wrong".

### Arduino CLI version

nightly

### Operating system

N/A

### Operating system version

N/A

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

alessio-perugini, 2023-09-28:
The JSON schema should be generated starting from a well-known structure. Unfortunately, the current schema was hand-written. Not being automatically generated has two downsides: difficulty in adding new properties, and the correctness of the values. This makes the schema completely unreliable.
In the CLI, we cannot perform the configuration validation, leaving the door open at runtime issues that could be prevented in the first place.

At this point, we have to rethink the configuration system from the ground up.

1. Define a Golang struct that represents the configuration.
2. Use that Config struct to add `jsonschema` tags and use some go [library](https://github.com/invopop/jsonschema) to generate the schema.
3. Understand if we can remove `viper`.

### Pros
- Validate config file at startup aka no more runtime errors
- Configs are now typed
- Jsonschema is automatically generated
- We can lower the viper footprint or even completely remove it

### Cons
- Require a good amount of effort and planning

alessio-perugini, 2025-03-12:
After the settings refactor, and after thinking more carefully we want to allow `additionProperties`.
Strictly allowing only the keys we expect doesn't provide more benefit to the end user. I can think of only some downsides in these cases:
- one user shares a config with another that is running an older cli version
- the user downgraded the cli version

closed 2025-03-12 by alessio-perugini