arduino/arduino-cli #2227: #include "file.xyz" fails to find "file.xyz" in sketch directory, for unusual file extensions

jrobert-gh, 2023-06-25:
### Describe the problem

In IDE 2.1.0, `#include` with a filename in quotes (not angle brackets) and the file exists in the sketch directory but has an unusual extension such as `xyz`, the compilation fails. The error message given is:

Detecting libraries used...
/Users/jrobert/Projects/ArduinoWksp/sketch_jun25a/sketch_jun25a.ino:3:12: fatal error: Test.xyz: No such file or directory

This example fails:
`
void setup() {
  Serial.begin(115200);
  #include "Test.xyz"
 }

void loop() {

}
`

If, instead of just the filename, its full path is given, the error is avoided.

### To reproduce

Demo 1:
- Create a new sketch with the above code.
- Create local include file named "file.xyz" containing `Serial.println("Hello, World");
`
 and terminated with a new-line.
- Compile the sketch.
- Note the error message, as stated above.

Demo 2:
- In the .ino file, replace the include file's name with its full path.
- Repeat the compilation. Note that the compilation now succeeds.

Demo 3:
- Rename the include file from "file.xyz" to "file.h"
- In the .ino file's #include statement, make the same substitution ("file.xyz" -> "file.h").
- Repeat the compilation. Note the the compilation succeeds again.

### Expected behavior

All three demos should have compiled without error.

### Arduino IDE version

Arduino IDE 2.1.0

### Operating system

macOS

### Operating system version

12.6.7

### Additional context

IDE 1.8.19 passes the above tests correctly.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-25:
Hi @jrobert-gh. Thanks for your interest in this open source project.

This is normal and expected. It is mandatory to use the supported file extensions that are listed in the Arduino Sketch Specification:

https://arduino.github.io/arduino-cli/dev/sketch-specification/#additional-code-files

> IDE 1.8.19 passes the above tests correctly.

I'm not sure I understand this statement but in case it is misunderstood I should state that you will get the same result in Arduino IDE 1.8.19 as in Arduino IDE 2.1.0 and Arduino CLI if you perform the steps described in the "To reproduce" section of the issue report, so this is not a matter of a change in behavior.

closed 2023-06-25 by per1234

jrobert-gh, 2023-06-25:
I did, in fact, run the same tests with v1.8.19, and got what I believe to be the proper results, as obtained by standard C/C++ preprocessors. To the extent that the often-called "Arduino Language" is standard C++, the behavior of the its preprocessor ought to reflect that as well.

matthijskooijman, 2023-06-26:
For a bit more background: What the Arduino build system does, is it preprocesses .ino files into .cpp files, and has gcc compile those. Because it does not want to write new files into your sketch directory, the generated files are written to a temporary directory. Due to limitations in gcc, the only way (AFAICT) to ensure that you can include header files from your sketch just like normal, is to copy the header files into the temporary directory. To prevent having to copy very big files (i.e. maybe you have some big PDFs or videos or whatever in your sketch directory), it only copies known file extensions, as indicated in the link by @per1234.

> To the extent that the often-called "Arduino Language" is standard C++, the behavior of the its preprocessor ought to reflect that as well.

I agree that this should be (and I think is) the goal, but it's just not always possible. I looked into gcc options for include paths, to see if this could somehow be made to work to point gcc at the original sketch directory for includes, but I think it turned out it was not possible to get exactly the right semantics - includes from the current directory are a bit special, I recall.