arduino/arduino-cli #2228: how to use `utility/` folder in library development

RodrigoDornelles, 2023-06-26:
I can't find clear documentation, I would like to put optional extras in my lib in addition to the main core.

 * **structfolder**
```
src/
  foo/
    foo.c
    foo.h
   libname.h
   libname.c
utility/
   libnameComplement.h
   libnameComplement.h
```

how to include `libnameComplement` ?

I'm using this link to guide me, but I'm not able to resolve the headers.
<https://arduino.github.io/arduino-cli/0.33/library-specification/>

```c
#include <libname.h> // works
#include <libnameComplement.h> // not works

void setup()
{
  libnamee_func();
  libnameComplement_func();
{

void loop()
{

}
```

per1234, 2023-06-26:
Hi @RodrigoDornelles. The key is this statement in the Arduino Library Specification:

https://arduino.github.io/arduino-cli/0.33/library-specification/#source-code

> For backward compatibility with Arduino IDE 1.0.x, the library author may opt to place source code into the root folder, instead of the folder called `src`. In this case the 1.0 library format is applied and the source code is searched from the library root folder and the `utility` folder

Note the "**instead**". You must not use both `src` and `utility` folders in the library. It is an either/or thing.

If you use the "1.5" library format, the contents of the `src` subfolder are compiled recursively. So you are welcome to organize the code files into subfolders of the `src` folder. So you can separate these "optional extras" into a subfolder of the `src` folder if you like. You can even name that folder `utility` if you like.

I'll provide an example of a valid library structure:

```text
libname/
├── library.properties
└── src/
    ├── foo/
    │   ├── foo.c
    │   └── foo.h
    ├── libname.cpp
    ├── libname.h
    └── libnameComplement/
        └── libnameComplement.h
```

> how to include `libnameComplement`?

The library users can use the following `#include` directives in their code:

```cpp
#include <libname.h>  // A file from the root source folder must be #included first to cause the library to be added to the compiler search path.
#include <libnameComplement/libnameComplement.h>  // The path is relative to the root source folder
```

closed 2023-06-26 by per1234

RodrigoDornelles, 2023-06-30:
@per1234 I was looking for something that doesn't always compile, as it ends up facing conflicts in my project in the way the Arduino IDE/CLI works.

``` 
libname/
├── library.properties
└── src/
    ├── foo/
    │   ├── foo.c
    │   └── foo.h
    ├── libname.cpp
    ├── libname.h
    └── libnameComplement/
        └── libnameComplement.h
    └── libnameComplement2/
        └── libnameComplement2.h
```

let's suppose that **libnameComplement** and **libnameComplement2** have a function of the same name called `samefunc()` and I want to give the availability of using **libname** to use this `samefunc()` according to what was chosen by the user of the IDE.

## Temporary solution ##

currently I made some not very elegant solutions, which fail to compile this function with the library, to make it compile with sketch and be properly linked.
```c
#include <libname.h>
#include <libnameComplement/libnameComplement.c>  //nasty solution for including a .c file
```