arduino/arduino-cli #2240: Using `--library` flag does not compile sketches

aliphys, 2023-07-07:
### Describe the problem

I cannot compile a sketch when providing a custom library location with the `--library` flag.

### To reproduce

1. Clone the nicla-sense-me-fw repo and open a terminal in the top most directory within VS Code
2. Run the command `arduino-cli compile --fqbn arduino:mbed_nicla:nicla_sense --library "C:\Users\USERNAME\Documents\GitHub\nicla-sense-me-fw\Arduino_BHY2"  "C:\Users\USERNAME\Documents\GitHub\nicla-sense-me-fw\Arduino_BHY2\examples\Standalone\Standalone.ino" --verbose`
3. After over ten minutes, the compilation does not complete. 

### Expected behavior

I can compile the sketch with a custom library without any problems. Or, if there is any a helpful message is shown. 

### Arduino CLI version

0,33,1

### Operating system

Windows

### Operating system version

Windows 11

### Additional context

- I have used similar syntax in previous versions and it worked. See: https://github.com/alrvid/randomcamerastuff/blob/main/camera.sh
- Without the `--verbose` nothing is printed to the terminal console.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

aliphys, 2023-07-07:
Here is an example of the output.
<details>
  <summary>
   Terminal output 
</summary>

```
PS C:\Users\Ali Jahangiri\Documents\GitHub\nicla-sense-me-fw> arduino-cli compile --fqbn arduino:mbed_nicla:nicla_sense --library "C:\Users\Ali Jahangiri\Documents\GitHub\nicla-sense-me-fw\Arduino_BHY2"  "C:\Users\Ali Jahangiri\Documents\GitHub\nicla-sense-me-fw\Arduino_BHY2\examples\Standalone\Standalone.ino" --verbose
FQBN: arduino:mbed_nicla:nicla_sense
Using board 'nicla_sense' from platform in folder: C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\4.0.2
Using core 'arduino' from platform in folder: C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\4.0.2

Detecting libraries used...
"C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/defines.txt" "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/cxxflags.txt" -DARDUINO_ARCH_NRF52832 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NICLA -DARDUINO_ARCH_MBED_NICLA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino" "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/includes.txt" "C:\\Users\\Ali Jahangiri\\AppData\\Local\\Temp\\arduino\\sketches\\AB81881D0BBA815140ABB01D519F2C7B\\sketch\\Standalone.ino.cpp" -o nul
Alternatives for Arduino_BHY2.h: [Arduino_BHY2@1.0.6]
ResolveLibrary(Arduino_BHY2.h)
  -> candidates: [Arduino_BHY2@1.0.6]
"C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/defines.txt" "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/cxxflags.txt" -DARDUINO_ARCH_NRF52832 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NICLA -DARDUINO_ARCH_MBED_NICLA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA" "-IC:\\Users\\Ali Jahangiri\\Documents\\GitHub\\nicla-sense-me-fw\\Arduino_BHY2\\src" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino" "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/includes.txt" "C:\\Users\\Ali Jahangiri\\AppData\\Local\\Temp\\arduino\\sketches\\AB81881D0BBA815140ABB01D519F2C7B\\sketch\\Standalone.ino.cpp" -o nul
Alternatives for ArduinoBLE.h: [ArduinoBLE@1.3.4]
ResolveLibrary(ArduinoBLE.h)
  -> candidates: [ArduinoBLE@1.3.4]
"C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/defines.txt" "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/cxxflags.txt" -DARDUINO_ARCH_NRF52832 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NICLA -DARDUINO_ARCH_MBED_NICLA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA" "-IC:\\Users\\Ali Jahangiri\\Documents\\GitHub\\nicla-sense-me-fw\\Arduino_BHY2\\src" "-IC:\\Users\\Ali Jahangiri\\Documents\\Arduino\\libraries\\ArduinoBLE\\src" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\Ali Jahangir [... cut ...] ata\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\cores\\arduino" "@C:\\Users\\Ali Jahangiri\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.2\\variants\\NICLA/includes.txt" "C:\\Users\\Ali Jahangiri\\Documents\\Arduino\\libraries\\ArduinoBLE\\src\\utility\\HCICordioTransport.cpp" -o nul
```

</details>

aliphys, 2023-07-07:
Related issues:
- https://github.com/arduino/arduino-cli/issues/1955
- https://github.com/arduino/arduino-cli/issues/2180
- https://github.com/arduino/arduino-cli/issues/2021

MatteoPologruto, 2023-09-25:
Hello @aliphys, I was not able to reproduce your issue. Here are the commands I ran.
```
arduino-cli version                                                                                            
arduino-cli.exe  Version: 0.33.1 Commit: 347bfeb0 Date: 2023-06-30T16:14:15Z
```
```
arduino-cli compile --fqbn arduino:mbed_nicla:nicla_sense --library "C:\Users\m.pologruto\Desktop\Arduino\nicla-sense-me-fw\Arduino_BHY2" "C:\Users\m.pologruto\Desktop\Arduino\nicla-sense-me-fw\Arduino_BHY2\examples\Standalone\Standalone.ino" --verbose
```
<details><summary>Output</summary>
FQBN: arduino:mbed_nicla:nicla_sense
Using board 'nicla_sense' from platform in folder: C:\Users\m.pologruto\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\4.0.6
Using core 'arduino' from platform in folder: C:\Users\m.pologruto\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\4.0.6

Detecting libraries used...
"C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/defines.txt" "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/cxxflags.txt" -DARDUINO_ARCH_NRF52832 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NICLA -DARDUINO_ARCH_MBED_NICLA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino" "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/includes.txt" "C:\\Users\\m.pologruto\\AppData\\Local\\Temp\\arduino\\sketches\\B92267E2148E73B698E8B0214B3F6D45\\sketch\\Standalone.ino.cpp" -o nul
Alternatives for Arduino_BHY2.h: [Arduino_BHY2@1.0.6 Arduino_BHY2@1.0.6]
ResolveLibrary(Arduino_BHY2.h)
  -> candidates: [Arduino_BHY2@1.0.6 Arduino_BHY2@1.0.6]
"C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/defines.txt" "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/cxxflags.txt" -DARDUINO_ARCH_NRF52832 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NICLA -DARDUINO_ARCH_MBED_NICLA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA" "-IC:\\Users\\m.pologruto\\Desktop\\Arduino\\nicla-sense-me-fw\\Arduino_BHY2\\src" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino" "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/includes.txt" "C:\\Users\\m.pologruto\\AppData\\Local\\Temp\\arduino\\sketches\\B92267E2148E73B698E8B0214B3F6D45\\sketch\\Standalone.ino.cpp" -o nul
Alternatives for ArduinoBLE.h: [ArduinoBLE@1.3.6]
ResolveLibrary(ArduinoBLE.h)
  -> candidates: [ArduinoBLE@1.3.6]
"C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/defines.txt" "@C:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA/cxxflags.txt" -DARDUINO_ARCH_NRF52832 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NICLA -DARDUINO_ARCH_MBED_NICLA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\variants\\NICLA" "-IC:\\Users\\m.pologruto\\Desktop\\Arduino\\nicla-sense-me-fw\\Arduino_BHY2\\src" "-IC:\\Users\\m.pologruto\\Documents\\Arduino\\libraries\\ArduinoBLE\\src" "-IC:\\Users\\m.pologruto\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nicla\\4.0.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\m.pologruto\\AppData\\Lo [... cut ...]
arduino:mbed_nicla 4.0.6   C:\Users\m.pologruto\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\4.0.6
</details>

Once I installed the required library `ArduinoBLE`, the compile ran smoothly using the correct custom library:
```
Multiple libraries were found for "Arduino_BHY2.h"
  Used: C:\Users\m.pologruto\Desktop\Arduino\nicla-sense-me-fw\Arduino_BHY2
  Not used: C:\Users\m.pologruto\Documents\Arduino\libraries\Arduino_BHY2
```
The same happened using the [latest](https://github.com/arduino/arduino-cli/releases/tag/0.34.2) CLI version.

aliphys, 2023-10-20:
@MatteoPologruto Can confirm that this issue is resolved in 2661f5d9 (0.32.2).

closed 2023-10-20 by aliphys