arduino/arduino-cli #2252: ESP32 flags unused function as error (should be warning)

vzahradnik, 2023-07-30:
### Describe the problem

The ACS712 library has a function static uint16_t _internalAnalog(uint8_t pin) that is not used by the examples.

The build sees this function is not used and generates an error on this, imho this should be a warning,
All other platforms tested do not report it as error.

Flag: -Werror=unused-function

build
https://github.com/RobTillaart/ACS712/actions/runs/3924992021/jobs/6747518244

### To reproduce

This may happen with various libraries. I encountered the same behavior with OneWire as a test dependency.
You can compile the following sketch:
https://github.com/JSC-electronics/acdu-support-library/blob/main/examples/TemperatureSensor/TemperatureSensor.ino

Required dependencies are listed here:
https://github.com/JSC-electronics/acdu-support-library/blob/main/.arduino-ci.yaml

### Expected behavior

Warnings are treated as warnings instead of errors. Especially when the unused variables are defined like this:

```cpp
__attribute__((unused)) volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
```

### Arduino CLI version

0.29.0

### Operating system

Linux

### Operating system version

ubuntu-latest on GitHub Actions

### Additional context

Related issue in the Arduino-CI repository:
https://github.com/Arduino-CI/arduino_ci/issues/345

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-07-31:
>imho this should be a warning

You're entitled to your own opinion, but the opinion of the ESP32 boards platform developers was that it should be an error, and so they configured the platform to add the `-Wno-error=unused-function` to the compilation commands that are generated by Arduino CLI:

https://github.com/espressif/arduino-esp32/blob/master/platform.txt

The [Arduino boards platform framework](https://arduino.github.io/arduino-cli/dev/platform-specification/) allows each platform developer great flexibility in configuring the behavior of the platform.

Since your report is not related to a defect in Arduino CLI, I'll close this as invalid. If you would like assistance with fixing the bug in the library code, please make a post on the Arduino Forum:

https://forum.arduino.cc

I'm sure we will be able to help you out over there.

closed 2023-07-31 by per1234

vzahradnik, 2023-07-31:
OK, thanks for the response anyway.