arduino/arduino-cli #2269: Remove `Port#hardware_id` from the gRPC API

kittaakos, 2023-08-18:
### Describe the request

As described [here](https://github.com/arduino/arduino-cli/issues/2245#issuecomment-1653110594), the `hardware_id` was introduced as a workaround to help CLI consumers to detect a potential port change during the upload.

> Once the feature is available and the CLI can provide the new port (if it has changed) after the upload, I propose eliminating the `hardwareId` from the gRPC API before the 1.0 release. I do not see why this should be exposed to the public. See the downstream change at [arduino/arduino-ide#1913](https://github.com/arduino/arduino-ide/pull/1913):
> 
> From [here](https://github.com/arduino/arduino-ide/pull/1913#issuecomment-1653105390):
> 
> > This PR has been created to support the "auto-reconnecting" to a board in DFU mode, which the R4 Minima goes into during upload and reset. The primary motivation was to make IDE2 compatible with the Uno R4 board.
> 
> Related API:
> 
> https://github.com/arduino/arduino-cli/blob/62bb4c47cb2bb64f642cba050a958f45902202ee/rpc/cc/arduino/cli/commands/v1/port.proto#L34-L35
> 
> I can open a separate issue if it's needed.

### Describe the current behavior

The `Port#hardware_id` is exposed.

### Arduino CLI version

38479dc706f27b2640cabad0bf39cd24e4f049c7

### Operating system

macOS

### Operating system version

13.5

### Additional context

Follow-up of https://github.com/arduino/arduino-cli/issues/2245.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

umbynos, 2023-09-01:
This is not required anymore for detection, but could be useful in other way. For example, for grouping ports exposed by the same board (not a use case we are currently supporting, but it's a possibility for the consumers)

kittaakos, 2023-09-01:
> For example, for grouping ports exposed by the same board (not a use case we are currently supporting, but it's a possibility for the consumers)

The API was added as a workaround. You don't need to keep it for made-up future use cases. If you decide to keep it, please properly document what clients are supposed to do with the `Port#hardware_id` for an upload request, for example.

I understand that you consider this API as something extra the CLI provides. As a consumer of the CLI, I need help understanding what to do with it.