arduino/arduino-cli #2274: [AE-127] Error message when udev code not configured on linux is unhelpful

aliphys, 2023-08-23:
### Describe the problem

It is required to configure the udev code in Linux to provide access to the Serial port. Otherwise, it is not possible to upload sketches to the Arduino board.

Given that configuring the udev rules is essential, then most (all?) linux users will face this issue. For new users, this could be especially difficult to figure out. 

### To reproduce

1. Update the Arduino-cli via [brew](https://arduino.github.io/arduino-cli/0.33/installation/#install-via-homebrew-macoslinux).
    ```
    brew update
    brew upgrade arduino-cli
    ```
2. Install the core for the Nicla Sense ME
    ```
    arduino-cli core install arduino:mbed_nicla
    ```
3. Compile a sketch (using [this](https://github.com/arduino/nicla-sense-me-fw/tree/main/Arduino_BHY2/examples/BHYFirmwareUpdate) as an example)
    ```
    user@user-laptop:~/github/nicla-sense-me-fw$ arduino-cli compile --fqbn arduino:mbed_nicla:nicla_sense Arduino_BHY2/examples/BHYFirmwareUpdate
    Sketch uses 259384 bytes (49%) of program storage space. Maximum is 527616 bytes.
    Global variables use 10768 bytes (16%) of dynamic memory, leaving 53520 bytes for local variables. Maximum is 64288 bytes.

    Used platform      Version Path                                                            
    arduino:mbed_nicla 4.0.4   /home/user/.arduino15/packages/arduino/hardware/mbed_nicla/4.0.4
    ```
4. Identify the serial port for the connected Nicla Sense ME board
    ```
    user@user-laptop:~/github/nicla-sense-me-fw$ arduino-cli board list
    Port         Protocol Type              Board Name             FQBN                           Core              
    /dev/ttyACM0 serial   Serial Port (USB) Arduino Nicla Sense ME arduino:mbed_nicla:nicla_sense arduino:mbed_nicla
    ```
5. Upload to the Nicla Sense ME board on the `/dev/ttyACM0` port. It is not possible to upload, and we get the `exit status 1 error`
    ```
    user@user-laptop:~/github/nicla-sense-me-fw$ arduino-cli upload --port /dev/ttyACM0 --fqbn arduino:mbed_nicla:nicla_sense Arduino_BHY2/examples/BHYFirmwareUpdate
    Open On-Chip Debugger 0.11.0+dev-gab95bac57-dirty (2021-05-11-10:49)
    Licensed under GNU GPL v2
    For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
    debug_level: 0adapter speed: 1000 kHzError: unable to open CMSIS-DAP device 0x2341:0x60
    Error: unable to find a matching CMSIS-DAP deviceFailed uploading: uploading error: exit status 1
    ```
6. We apply the udev rules as stated at the bottom of [this](https://docs.arduino.cc/software/ide-v2/tutorials/getting-started/ide-v2-downloading-and-installing) page.
7. It is now possible to upload the sketch
    ```
    user@user-laptop:~/github/nicla-sense-me-fw$ arduino-cli upload --port /dev/ttyACM0 --fqbn arduino:mbed_nicla:nicla_sense Arduino_BHY2/examples/BHYFirmwareUpdate
    Open On-Chip Debugger 0.11.0+dev-gab95bac57-dirty (2021-05-11-10:49)
    Licensed under GNU GPL v2
    For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
    debug_level: 0

    adapter speed: 1000 kHz

    target halted due to debug-request, current mode: Thread 
    xPSR: 0x01000000 pc: 0x00006dd0 msp: 0x20010000
    target halted due to debug-request, current mode: Thread 
    xPSR: 0x01000000 pc: 0x00006dd0 msp: 0x20010000
    ** Programming Started **
    ** Programming Finished **
    shutdown command invoked
    New upload port: /dev/ttyACM0 (serial)
    ```

### Expected behavior

The user is informed of this issue when error occur. Something like `If using a Linux OS, make sure that the udev rules are configured correctly to enable access to the Serial port. See https://support.arduino.cc/hc/en-us/articles/9005041052444-Fix-udev-rules-on-Linux for more information`

### Arduino CLI version

0.34

### Operating system

Linux

### Operating system version

Ubuntu 22.04.03 LTS

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

aliphys, 2023-08-23:
Related issues:
- https://github.com/arduino/arduino-cli/issues/1084
- https://github.com/arduino/arduino-cli/issues/645

Related forum post:
- https://forum.arduino.cc/t/unable-to-upload-sketch-with-ide-2-in-linux/1046973/6 

This [Help Centre](https://support.arduino.cc/hc/en-us/articles/9005041052444-Fix-udev-rules-on-Linux) article is very helpful. However, it does not show up high in google when searching for the error message.
[image]

umbynos, 2023-09-01:
We may allow a platform to provide a way to augment the error messages of the uploader, similarly to what's described in  https://github.com/arduino/arduino-cli/issues/1657