arduino/arduino-cli #2304: Runtime error using `arduino-cli upload` with version 0.34

automatiek, 2023-09-08:
### Describe the problem

Trying to upload a binary to the Arduino Nano BLE 33 Sense on version 0.34 of the cli tools fails with following output:
```
shawnhymel@Shawn-Hymel-LQNW70Y61L arduino-cli_0.34.1_macOS_ARM64 % ./arduino-cli upload -p /dev/cu.usbmodem1201 --fqbn arduino:mbed_nano:nano33ble --input-dir .
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x2 addr=0x68 pc=0x102f7df24]

goroutine 1 [running]:
github.com/arduino/arduino-cli/arduino/sketch.(*Sketch).GetDefaultFQBN(...)
	/home/build/arduino/sketch/sketch.go:241
github.com/arduino/arduino-cli/internal/cli/upload.runUploadCommand(0x14000413400?, {0x1400019af00, 0x0, 0x6?})
	/home/build/internal/cli/upload/upload.go:103 +0x1d4
github.com/spf13/cobra.(*Command).execute(0x14000413400, {0x1400019ae40, 0x6, 0x6})
	/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:860 +0x4a8
github.com/spf13/cobra.(*Command).ExecuteC(0x14000313680)
	/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x338
github.com/spf13/cobra.(*Command).Execute(0x0?)
	/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:902 +0x1c
main.main()
	/home/build/main.go:31 +0x110
```

### To reproduce

Use this command to reproduce
```
arduino-cli_0.34.1_macOS_ARM64 % ./arduino-cli upload -p /dev/cu.usbmodem1201 --fqbn arduino:mbed_nano:nano33ble --input-dir .
```

### Expected behavior

Output using version 0.33.1:

```
shawnhymel@Shawn-Hymel-LQNW70Y61L arduino-cli_0.33.1_macOS_ARM64 % ./arduino-cli upload -p /dev/cu.usbmodem1201 --fqbn arduino:mbed_nano:nano33ble --input-dir .
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Erase flash

Done in 0.001 seconds
Write 262200 bytes to flash (65 pages)
[==============================] 100% (65/65 pages)
Done in 10.397 seconds
```

### Arduino CLI version

0.34.1

### Operating system

macOS

### Operating system version

Ventura 13.5 on Apple M1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2023-09-08:
I can reproduce and confirm this is a regression on MacOSX.

cmaglie, 2023-09-08:
Also, this seems already fixed on `master`, I guess I can cherry-pick the commits and release a patched version `0.34.2`

automatiek, 2023-09-08:
thanks @cmaglie for picking this up so rapidly.

cmaglie, 2023-09-11:
Fixed in version `0.34.2`.

closed 2023-09-11 by cmaglie