arduino/arduino-cli #2323: Compilation error:  No such file or directory

3safiber, 2023-09-16:
### Describe the problem

I cannot use any library that I download

### To reproduce

I downloaded the library ArduinoJson  then write the code:

```
#include <ArduinoJson.h>
void setup() {
}
void loop() {
}
```

### Expected behavior

[image]

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-09-16:
Hi @3safiber. Thanks for your report. I'm going to ask you for some additional information.

Please do this:

1. Select **File > Preferences...** from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Copy the full path from the "**Sketchbook location**" field of the "**Preferences**" dialog.
1. Paste the copied path in a comment here on the GitHub issue.

3safiber, 2023-09-19:
Hi @per1234 , this is the full path :
c:\Users\salam\OneDrive\المستندات\Arduino