arduino/arduino-cli #2334: Nano33BLE Serial port is not available right after upload

ManjunathaVG-MathWorks, 2023-09-22:
### Describe the problem

Hi,
I am currently working on a PowerShell script that involves uploading a file to a Nano33BLE board using the command-line interface (CLI). Once the upload is complete, I need to open the serial port of the board and send some commands.

During my testing, I observed that the script only works properly if there is a delay between the upload command and the serial port open command.

I have attached logs for both scenarios, one with the delay and one without. My question is whether adding this delay is necessary or if it indicates a bug in the script?

### To reproduce

[logs.txt](https://github.com/arduino/arduino-cli/files/12700982/logs.txt)

Powershell script
***************************************************************************************************************
C:\24a\aCLI\arduino-cli -v --fqbn arduino:mbed:nano33ble compile --upload "C:\Users\mvankada\AppData\Local\Temp\ArduinoServerNano33BLE" --port COM15 --build-path "C:\Users\mvankada\AppData\Local\Temp\ArduinoServerNano33BLE\MW" --config-file "C:\24a\aCLI\arduino-cli.yaml"

#Start-Sleep -Seconds 10
$port= new-Object System.IO.Ports.SerialPort COM15,115200,None,8,one

$port.open()
#$port.WriteLine("some string")
#$port.ReadLine()
#$port.Close()
*******************************************************************************************************************

### Expected behavior

Serial port should be free/available right after the upload command

### Arduino CLI version

0.33.0

### Operating system

Windows

### Operating system version

Windows 11 (OS Build 22621.2283)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2023-09-25:
Could you retry with the CLI version 0.34.2? We implemented an algorithm that should wait for the upload port to be available after the upload.

cmaglie, 2023-09-27:
In the second run it's failing the upload:
```
Performing 1200-bps touch reset on serial port COM15
Waiting for upload port...
No upload port found, using COM15 as fallback
"C:\Users\mvankada\Downloads\new\data\packages\arduino\tools\bossac\1.9.1-arduino2/bossac.exe" -d --port=COM15 -U -i -e -w "C:\Users\mvankada\AppData\Local\Temp\ArduinoServerNano33BLE\MW/ArduinoServerNano33BLE.ino.bin" -R
No device found on COM15
Error during Upload: Failed uploading: uploading error: exit status 1      <----------------
Exception calling "Open" with "0" argument(s): "Access to the port 'COM15' is denied."
At C:\Users\mvankada\Downloads\new\mbedcli.ps1:7 char:1
+ $port.open()
+ ~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : UnauthorizedAccessException
```

It seems that the board is not getting the "1200-bps touch reset" to start the bootloader, so in this case, the issue is not due to the waiting after the upload, but it's the upload itself that is failing. From time to time I had similar issues, I noticed that the firmware is very sensitive to timings, something you can try:
* double-check that the COM port is not in use by other running processes
* replace the USB cable
* change USB port on the PC
* remove/add a USB-HUB

Also, adding a check on the return value of the `arduino-cli` upload may help debug. A quick and dirty workaround may be to retry the upload if the first one fails...

ManjunathaVG-MathWorks, 2023-10-03:
Thank you for your reply.

I have followed all the steps you mentioned, but I am still encountering the same issue. I am not sure whether the problem lies with the board. For now, I am using retry logic which seems to be resolving the problem.

closed 2023-10-03 by ManjunathaVG-MathWorks