arduino/arduino-cli #2341: Release Arduino CLI with `v` prefix

umbynos, 2023-09-26:
We're using some packages of the arduino-cli in other projects. Like fw-uploader plugins, arduino-create-agent and more.
When you don't tag with the v prefix golang treat that tag as it's a pseudo-version and import like: `github.com/arduino/arduino-cli v0.0.0-20230529151818-ca60d4b47be4`
More info here: https://go.dev/ref/mod#versions

alessio-perugini, 2023-09-26:
One gotcha we have to think of:
- IDE2 assumes no v prefix. (We have to coordinate with them).
- We have to investigate if our tools published in the packing index without the `v` prefix still work or if we need to add the `v` prefix in the index as well.

umbynos, 2023-09-26:
We have also to verify this will continue to work https://github.com/Homebrew/homebrew-core/blob/e866529e869478aac95e975dad21991c7dd2912e/Formula/a/arduino-cli.rb

PR #2374 (2023-10-19): Introduce new "v" version format
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [X] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING [... cut]
closed 2023-10-19 by umbynos