arduino/arduino-cli #2352: feature: add some specific exit error-codes for `daemon` startup

cmaglie, 2023-10-02:
### Describe the request

From @kittaakos [comment](https://github.com/arduino/arduino-cli/issues/1762#issuecomment-1354512737)

> Besides the application-level error handling for the services, I would like to get a proper exit code and a terminating daemon process when something went wrong. Now, IDE2 must sniff into the stderr of the process, parse such messages and kill the daemon process because it's not terminating, although the daemon is nonfunctional:
>
>  - To determine a `Failed to listen on TCP port. Address already in use.` error message:
>    `failed to listen`, `address already in use`, `bind`,  `only one usage of each socket address`
>
>  - For invalid address: `Failed to listen on TCP port. Unknown address.`:
>    `is unknown name` or `tcp/` and `is an invalid port`
>
>  - To provide the `Failed to listen on TCP port. Invalid port.` to the user:
>    `is an invalid port` and `failed to serve`

### Describe the current behavior

Arduino CLI daemon will fail with no specific exit code.

### Arduino CLI version

nightly

### Operating system

N/A

### Operating system version

N/A

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

cmaglie, 2023-10-02:
It also seems to fail if an invalid data-dir config is present:
```
% ./arduino-cli version                                 
arduino-cli  Version: git-snapshot Commit: c506f6a4 Date: 2023-10-02T15:09:22Z
% cat ./arduino-cli.yaml                                    
directories:
  data: /this/will/fail/on/macon/it/cannot/create/root/folders
% ./arduino-cli daemon --port 0 --config-file ./arduino-cli.yaml
Error: invalid path creating config dir: /this/will/fail/on/macon/it/cannot/create/root/folders error: mkdir /this: read-only file system
```

kittaakos, 2023-11-03:
https://github.com/arduino/arduino-cli/issues/2352#issuecomment-1743234007 is not yet resolved via https://github.com/arduino/arduino-cli/pull/2400.

closed 2023-11-06 by cmaglie