arduino/arduino-cli #2353: Error during install: Error downloading tool esp32:openocd-esp32@v0.11.0-esp32-20221026: no versions available for the current OS

lonebots, 2023-10-02:
### Describe the problem

I was trying to install `esp32` board to my machine using the `arduino-cli` and this was the result 

```bash
$ arduino-cli core install esp32:esp32
Downloading packages...
esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:xtensa-esp-elf-gdb@11.2_20220823 esp32:xtensa-esp-elf-gdb@11.2_20220823 already downloaded
esp32:riscv32-esp-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:riscv32-esp-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:riscv32-esp-elf-gdb@11.2_20220823 esp32:riscv32-esp-elf-gdb@11.2_20220823 already downloaded
Error during install: Error downloading tool esp32:openocd-esp32@v0.11.0-esp32-20221026: no versions available for the current OS, try contacting hristo@espressif.com

```

### To reproduce

1. install `arduino-cli`
2. run the following command `arduino-cli core install esp32:esp32`

### Expected behavior

The expected behavior is the cli tool to properly installed the requested board. 

### Arduino CLI version

Version: 0.34.1 Commit: 048415c5 Date: 2023-08-31T11:46:44Z

### Operating system

Linux

### Operating system version

22.04.3 LTS (Jammy Jellyfish)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

alessio-perugini, 2023-10-02:
:thinking: That usually happens when there aren't any tools available that match your host architecture.
Do you mind sending the output of `uname -r` :pray:

expert-amateur, 2023-10-03:
> 🤔 That usually happens when there aren't any tools available that match your host architecture. Do you mind sending the output of `uname -r` ? 🙏

Hi, I have the same problem, here's my output for uname -r:
6.2.0-33-generic

Thank you!

alessio-perugini, 2023-10-03:
@expert-amateur my bad, I've noticed that I typed the wrong flag. Could you run `uname -a`? That would tell me among other info if you're running 32 or 64 bit.

lukew151, 2023-10-11:
Same here:
```
$ uname -a
Linux luke-NL40-50CU 5.15.0-84-generic #93-Ubuntu SMP Tue Sep 5 17:16:10 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ arduino-cli core install esp32:esp32
Downloading packages...
esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:xtensa-esp-elf-gdb@11.2_20220823 esp32:xtensa-esp-elf-gdb@11.2_20220823 already downloaded
esp32:riscv32-esp-elf-gcc@esp-2021r2-patch5-8.4.0 esp32:riscv32-esp-elf-gcc@esp-2021r2-patch5-8.4.0 already downloaded
esp32:riscv32-esp-elf-gdb@11.2_20220823 esp32:riscv32-esp-elf-gdb@11.2_20220823 already downloaded
Error during install: Error downloading tool esp32:openocd-esp32@v0.11.0-esp32-20221026: no versions available for the current OS, try contacting hristo@espressif.com

$ arduino-cli version
arduino-cli  Version: 0.34.2 Commit: 963c1a76 Date: 2023-09-11T10:05:41Z
```

I'm running Ubuntu Jammy Jellyfish on a intel i7 laptop.

lonebots, 2023-10-19:
This is my result for `uname -a` 
```bash
Linux Legion 6.2.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Oct  6 10:23:26 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

I think this is the issue when, trying to install arduino-cli from the snap. I got the issue rectified after following the [official documentation](https://arduino.github.io/arduino-cli/0.35/installation/) for installing `arduino-cli`

umbynos, 2023-10-26:
There are some problems with snap installation. I think the problem lies on the `strict` confinement: https://snapcraft.io/docs/snap-confinement used [here](
https://github.com/manchoz/arduino-cli-snap/blob/855131b936d7be218fd5a024c66c3b5747ad6802/snap/snapcraft.yaml#L28

The cli installed with snap does not use normal `~/.arduino15` dir but a confined `~/snap/arduino-cli/43/.arduino15`