arduino/arduino-cli #2365: esp32_https_server doesnt work with arduino-cli

zerofo, 2023-10-13:
### Describe the problem

i using dockerfile to make a env , build the esp32_https_server examples. but it doesnt work with arduino-cli( it can run with arduino IDE )

### To reproduce

https://github.com/fhessel/esp32_https_server/issues/176

### Expected behavior

the firmware doesnt work.

### Arduino CLI version

0.34.2

### Operating system

Linux

### Operating system version

ubuntu22.04

### Additional context

https://github.com/fhessel/esp32_https_server/issues/176

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-10-13:
Hi @zerofo. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2023-10-13 by per1234

zerofo, 2023-10-13:
> 嗨 .感谢您对这个开源项目的关注。此问题跟踪器仅用于报告特定于项目的错误或功能请求。本主题更适合 Arduino 论坛。我相信我们将能够在那里为您提供帮助：
> 
> https://forum.arduino.cc/

but... it only cant work with arduino-cli...