arduino/arduino-cli #2368: The filename or extension is too long

luftaquila, 2023-10-13:
### Describe the problem

In Windows, I cannot compile a sketch with a following error.

```
Used platform Version Path
esp32:esp32   2.0.14  C:\Users\LUFT-AQUILA\local\workspace\monolith\builder\toolchain\arduino\data\packages\esp32\hardware\esp32\2.0.14
Error during build: fork/exec C:\Users\LUFT-AQUILA\local\workspace\monolith\builder\toolchain\arduino\data\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++.exe: The filename or extension is too long.
```

It seems like that `xtensa-esp32-elf-g++.exe: The filename or extension is too long.` is the problem.

I found some related issues #839 and #961.

I think this problem is fixed at around `0.14` according to the issues above, but I'm experiencing the problem in `0.34`.

`platform.local.txt` at https://github.com/arduino/arduino-cli/issues/839#issuecomment-714525612 did not worked.

### To reproduce

```
arduino-cli --config-file ./config/arduino-cli.yaml compile --board-options LoopCore=0 -e --fqbn esp32:esp32:esp32 -v ../device/telemetry/telemetry.ino
```

The default ESP32 example sketches instead of mine have same problem, too.

`arduino-cli.yaml`

```yaml
board_manager:
  additional_urls:
  - https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
build_cache:
  compilations_before_purge: 10
  ttl: 720h0m0s
daemon:
  port: "50051"
directories:
  data: ./toolchain/arduino\data
  downloads: ./toolchain/arduino\downloads
  user: ./toolchain/arduino\user
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
output:
  no_color: false
sketch:
  always_export_binaries: false
updater:
  enable_notification: true
```

### Expected behavior

A normal compilation as IDE does, instead of `The filename or extension is too long` error.

[Arduino IDE 2.2.1 compile output](https://pastebin.com/TWM3KMcT)

[arduino-cli 0.34 compile output](https://pastebin.com/jPpQXcs4)

### Arduino CLI version

arduino-cli.exe  Version: nightly-20231013 Commit: 3a2f9f1 Date: 2023-10-13T01:28:47Z

### Operating system

Windows

### Operating system version

Windows 11 Pro 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

luftaquila, 2023-10-14:
I found out that compile command length for the `Generating function prototypes...` was 32798 byte, which slightly exceeded 32768 Windows command length limit.

This was much longer than the IDE generates, which is 27206 bytes. And if I truncate all data path from both commands, there length were almost same(18671 for IDE, 18282 for arduino-cli). The major difference was just a data path length.

I moved the project directory directly under C:\ and it compiles normally.

BTW, I think this problem needs some robust solution instead of depending on absolute path length of `directories.data` config.

I didn't take a closer look at it, but I thought https://github.com/arduino/arduino-cli/issues/839#issuecomment-714525612 was a solution for diminishing commands length. But it didn't worked at all, including insert content directly into `platform.txt`.

pchemguy, 2024-10-16:
I have the same problem with both Arduino Legacy IDE and Arduino IDE V2 on Windows 10 when using the latest
"Arduino ESP32 Boards" 2.0.18-20240930.arduino3. With "Arduino ESP32 Boards" 2.0.13 everything works just fine.