arduino/arduino-cli #2370: Failed to install `arduino-cli` using the script

LRDPRDX, 2023-10-17:
### Describe the problem

Sorry if there is an issue regarding this, I didn't find any.

I am trying to install `arduino-cli` on raspberry pi 4.

```
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=~/local/bin sh
```

and I'm getting the error:

```
Installing in /home/raspy/local/bin
ARCH=ARM64
OS=Linux
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.34.2_Linux_ARM64.tar.gz
tar: /tmp/arduino-cli_0.34.2_Linux_ARM64.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
Failed to install arduino-cli
```

### To reproduce

```
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=~/local/bin sh
```

### Expected behavior

`arduino-cli` should be installed.

### Arduino CLI version

0.34.2 (trying to install)

### Operating system

Linux

### Operating system version

Ubuntu 23.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

tobozo, 2023-10-20:
quick fix: find the `checkLatestVersion()` function in the shell script and properly escape the offending `v?` prefix in the regular expression

before:

    CHECKLATESTVERSION_REGEX="v?[0-9][A-Za-z0-9\.-]*"

after:

    CHECKLATESTVERSION_REGEX="v\?[0-9][A-Za-z0-9\.-]*"

brentru, 2023-10-20:
@tobozo I modified the shell script locally and pointed a local fork of ci-arduino (which installs this script on a GH actions runner) at it. Works perfectly for now, thanks for the patch!

cmaglie, 2023-10-20:
Could you provide a PR for that?

brentru, 2023-10-20:
@cmaglie Implemented @tobozo 's solution in https://github.com/arduino/arduino-cli/pull/2380

per1234, 2023-10-20:
> quick fix: find the checkLatestVersion() function in the shell script and properly escape the offending v? prefix in the regular expression

Thanks @tobozo and @brentru! However, the regression introduced by https://github.com/arduino/arduino-cli/commit/74cdc80024357970cedcdb851e120faab196ab0b can not be the cause of the error reported by @LRDPRDX. The reason is that @LRDPRDX's report was made 2023-10-16 and the regression was introduced 2023-10-19.

So we can not consider this as resolved by https://github.com/arduino/arduino-cli/pull/2380

umbynos, 2023-12-01:
I just tried to install the CLI with the command above and it went smoothly on a raspberry PI. Are you sure you have the correct permissions for `/home/raspy/local/bin`?

LRDPRDX, 2023-12-02:
Hi, @umbynos . Actually I was able to reproduce this issue on my desktop. More over, I see your comment is after the fix commit has been merged.

LRDPRDX, 2023-12-02:
@umbynos Actually, now I just tried to reproduce it on my laptop -- everything works great. I will check the permissions etc on my raspberry on Monday (the device islocated at my working place).

alessio-perugini, 2024-01-04:
@LRDPRDX any update on this? :nerd_face:

cmaglie, 2024-10-28:
In the meantime, a lot of effort went into the `install.sh` script, so I'm closing this issue as outdated.
If you are still experiencing the issue, feel free to reopen it or, better yet, open a new issue so we can restart with fresh information.

closed 2024-10-28 by cmaglie