arduino/arduino-cli #2410: grpc: failed to unmarshal the received message: proto: bad wiretype

homiodev, 2023-11-08:
### Describe the problem

arduino-cli version = arduino-cli.exe  
    Version: 0.34.2 Commit: 963c1a76 Date: 2023-09-11T10:05:41Z
arduino-cli.exe core search arduino:avr 
   ID          Version Name
   arduino:avr 1.8.6   Arduino AVR Boards
Command: grpcui -max-msg-sz 600000000 -import-path ./rpc -proto cc/arduino/cli/commands/v1/commands.proto -plaintext localhost:50051
   MethodName: PlatformSearch
   instance: 1
   searchArg: arduino:avr
Response: Internal (13) grpc: failed to unmarshal the received message: proto: bad wiretype.
PS. Same answer from my java app

### To reproduce

Described in Descrive section

### Expected behavior

Result list return successfully

### Arduino CLI version

0.34.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

homiodev, 2023-11-08:
PS. Java app error:
detailMessage = "CANCELLED: Failed to read message."
cause = {StatusRuntimeException@30182} "io.grpc.StatusRuntimeException: INTERNAL: Invalid protobuf byte sequence"

cmaglie, 2023-11-08:
The .proto file definitions must match the version of the `arduino-cli` (always use the same protobuf definitions provided with the CLI), at least until version 1.0.0 is released.

Please be aware that breaking changes are expected in the gRPC until the 1.0.0 release, in particular, a lot of breaking changes are expected for the 0.36.0.

See https://github.com/arduino/arduino-cli/blob/master/docs/UPGRADING.md for details.

homiodev, 2023-11-12:
I see. Sad that api changes time to time

closed 2023-11-12 by homiodev