arduino/arduino-cli #2415: Avr-g++: no such file or directory

asimovc, 2023-11-10:
### Describe the problem

When i try to compile a simple Sketch i get this error `Error during build: fork/exec /home/hellboy/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: no such file or directory` and i already downloaded the avr core as you can see here:

```
~$ arduino-cli core list
ID          Installed Latest Name              
arduino:avr 1.8.6     1.8.6  Arduino AVR Boards
```
And the verbose says it:

```
FQBN: arduino:avr:mega
Using board 'mega' from platform in folder: /home/hellboy/.arduino15/packages/arduino/hardware/avr/1.8.6
Using core 'arduino' from platform in folder: /home/hellboy/.arduino15/packages/arduino/hardware/avr/1.8.6

Detecting libraries used...
/home/hellboy/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/hellboy/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/home/hellboy/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega /tmp/arduino/sketches/6A5A29022208ED9700955194B4CCD27C/sketch/Test.ino.cpp -o /dev/null

Used platform Version Path                                                        
arduino:avr   1.8.6   /home/hellboy/.arduino15/packages/arduino/hardware/avr/1.8.6
Error during build: fork/exec /home/hellboy/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: no such file or directory
```

### To reproduce

just use `arduino-cli compile -b arduino:avr:mega Sketch`

### Expected behavior

Compile the sketch

### Arduino CLI version

0.34.2 Commit: 963c1a76 Date: 2023-09-11T10:05:42Z

### Operating system

Linux

### Operating system version

Kiss Linux With Latest Version

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

umbynos, 2023-11-13:
Seems related to https://github.com/arduino/arduino-cli/issues/1208

cmaglie, 2024-01-04:
Does `/home/hellboy/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++` exists?

asimovc, 2024-03-28:
Turns out that is a problem related to glibc link, i used a musl distro when i have this bug. Just build and use gcompat solved the problem with the bin linked against glibc