arduino/arduino-cli #2465: CLI initialisation issues on Windows

Hannes7eicher, 2023-12-14:
### Describe the problem

When uploading this code snippet to the Nano ESP32 nothing gets printed when opening the serial monitor.

`
void setup() {
  Serial.begin(9600);
  while(!Serial);
  Serial.println("Starting...");
}

void loop() {}
`

When opening the serial monitor with [putty](https://www.putty.org/) the prompt is printed as expected.

### To reproduce

Upload a sketch to the Nano ESP32 either using the IDE or cli that uses `while(!serial)` and prints messages when opening the serial monitor. Open the serial monitor.

### Expected behavior

When opening the serial monitor I expect to see the messages printed inside `setup`.

### Arduino CLI version

0.34.2 Commit: 963c1a76 Date: 2023-09-11T10:05:41Z

### Operating system

Windows

### Operating system version

Windows 11 Pro

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2024-02-15:
@Hannes7eicher I see a bizarre behavior, the `Starting...` message appears if I open the port two times in a row:
```
C:\Users\cmagl\Desktop\Test>..\arduino-cli.exe monitor -p COM12
Connesso a COM12! Premere CTRL-C per uscire.
^C

C:\Users\cmagl\Desktop\Test>..\arduino-cli.exe monitor -p COM12
Connesso a COM12! Premere CTRL-C per uscire.
Starting...
```

On the second try the `Starting...` message is triggered as expected. This is not happening with `putty` that shows the correct behavior. Is this happening also with your system?

I cannot reproduce it on my Linux box either, so this is a Windows-related bug.

cmaglie, 2024-02-16:
The problem might be fixed with this patch: https://github.com/bugst/go-serial/pull/174

Xayton, 2024-02-19:
Need to release the new version of the serial-monitor in the package index.

closed 2024-03-07 by cmaglie