arduino/arduino-cli #2481: Support ARDUINO_CONFIG environment variable

weshouman, 2023-12-28:
### Describe the request

Is it possible to support setting the configuration file using an environment variable?

Currently it's possible to set the config file through multiple ways, like `--config-file` and by auto checking the current and parent directories. However, using an environment variable to ensure all the upcoming commands will be using the correct config file would ensure consistency and confidence in the result.

Note, as a current workaround, I'm shadowing the arduino-cli command using the following bash function, which indeed works, but it's not a clean solution.

```
arduino-cli() {
    if [ -n "$ARDUINO_CONFIG" ]; then
        # ARDUINO_CONFIG is set, use it with --config-file
        command arduino-cli --config-file "${ARDUINO_CONFIG}" "$@"
    else
        # ARDUINO_CONFIG is not set, call arduino-cli normally
        command arduino-cli "$@"
    fi
}
```

### Describe the current behavior

No environment variable are mentioned in the documentation that would set the --config-file 

### Arduino CLI version

Version: 0.34.2

### Operating system

Linux

### Operating system version

Any

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

alessio-perugini, 2024-01-02:
:wave: We currently support something similar via `ARDUINO_DIRECTORIES_DATA`. When that env var is set the cli will search for `arduino-cli.yaml` in the path specified by that env. [(if you're curious here is the snippet)](https://github.com/arduino/arduino-cli/blob/master/internal/cli/configuration/configuration.go#L50).

weshouman, 2024-01-03:
Great, thanks for the hint.

---

FYI, the confusion was due to the quoted [locations section in the docs](https://arduino.github.io/arduino-cli/0.35/configuration/#locations) , `as configured by directories.data`  was understood as only what's in the file, but not also the environment variable that is used for it.

```
Configuration files in the following locations are recognized by Arduino CLI:
    Location specified by the [--config-file](https://arduino.github.io/arduino-cli/0.35/commands/arduino-cli/) command line flag
    Arduino CLI data directory (as configured by directories.data)
```

weshouman, 2024-01-03:
This actually works.

However, I've just rechecked my use case, and I do have multiple configuration files in the same directory, each with a different name, thus this solution of specifying a directory for the arduino-cli.yaml does not allow arbitrary configuration file naming as allowed by `--config-file`

alessio-perugini, 2024-01-04:
Ok, we can add that.
Implementation notes:
- add env name: `ARDUINO_CONFIG_FILE`
- add doc

alessio-perugini, 2024-01-04:
Will be shipped in the `v0.36.0` or directly in the `v1.0.0-rc.1`

closed 2024-01-04 by alessio-perugini