arduino/arduino-cli #2483: Failing Github Actions CI using install.sh

tyeth, 2024-01-02:
### Describe the problem

all our CI is failing and I've traced through, and it seems like the grep is failing, but more like the newlines are not being detected by grep (or the $APPLICATION_DIST is incorrectly set - but the first arduino.cc download attempt says otherwise). Works in windows (ignoring cut), and wsl fine, but not in github actions.
CI Output: https://github.com/adafruit/Adafruit_Arcada/actions/runs/7388865095/job/20100744127#step:5:84
Which leads (via [our ci script](https://github.com/adafruit/ci-arduino/blob/master/actions_install.sh#L32)) to the grep listing more than one file here: https://github.com/arduino/arduino-cli/blob/master/install.sh#L171
 I have no idea really, escaping or maybe the environment variables in actions are causing problems as the arch is detected and the commands work apart from that ([there was some example of migrating to return stdout in a special github actions format instead](https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#example-of-writing-an-environment-variable-to-github_env)).

```
Installing in /home/runner/work/Adafruit_Arcada/Adafruit_Arcada/bin
ARCH=64bit
OS=Linux
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_v0.35.0_Linux_64bit.tar.gz
Did not find a release for your system: Linux 64bit
Trying to find a release using the GitHub API.
Downloading https://github.com/arduino/arduino-cli/releases/download/v0.35.0/0.35.0-checksums.txt
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_configuration.schema.json
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_32bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_64bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARM64.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARMv6.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARMv7.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_macOS_64bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_macOS_ARM64.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_proto.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_32bit.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_64bit.msi
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_64bit.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/CHANGELOG.md
Failed to install arduino-cli
Error: Process completed with exit code 3.
```

### To reproduce

having setup paths etc run
`curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh`

### Expected behavior

Script installs Linux_x64bit release from github

### Arduino CLI version

0.35.0

### Operating system

Linux

### Operating system version

23.04 maybe? github actions ubuntu-latest

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

tyeth, 2024-01-02:
At the advice of @brentru I'm tagging the people in the know... @cmaglie @facchinm @per1234 @alessio-perugini

ozanges, 2024-01-02:
Hi, 
The problem is, in my case : 
`$APPLICATION_DIST = arduino-cli_v0.35.0_Linux_64bit.tar.gz`

but we have 

```
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/0.35.0-checksums.txt
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_configuration.schema.json
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_32bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_64bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARM64.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARMv6.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARMv7.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_macOS_64bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_macOS_ARM64.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_proto.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_32bit.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_64bit.msi
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_64bit.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/CHANGELOG.md
```
for `echo "$BODY" | grep 'browser_' | cut -d\" -f4 `

So first var contains a **v** for **arduino-cli_v0.35** but non **v** for second var $DOWNLOAD_URL (arduino-cli_0.35...)

tyeth, 2024-01-02:
Not sure if it's okay to update the taskfile.yml to prepend v, as it seems the go routine getting the version via semver and drops the v prefix (skipping over quickly - not a go person). Alternatively the version.go file could be modified, or if undesired then the release-go-task.yml and publish-go-nightly-task.yml and publish-go-tester-task.yml could be modified instead.

ozanges, 2024-01-02:
Thanks @tyeth for the quick fix. I will test it when [PR](https://github.com/arduino/arduino-cli/pull/2484) will be validated.

cmaglie, 2024-01-03:
Thanks for the prompt bug report!

We have fixed the `install.sh` in `master` branch, the problem should be fixed now.
Please reopen or comment here if the issue is still present.

closed 2024-01-03 by cmaglie

ozanges, 2024-01-03:
Tested & fixed.
Thanks for your efficiency