arduino/arduino-cli #2486: Version 0.35.0 not installing on Jetson

yerachw, 2024-01-03:
### Describe the problem

Failed to install arduino-cli

nvidia@seeed:~$ ./install_arduino.sh
Installing in /home/nvidia/bin
ARCH=ARM64
OS=Linux
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_v0.35.0_Linux_ARM64.tar.gz
Did not find a release for your system: Linux ARM64
Trying to find a release using the GitHub API.
Downloading https://github.com/arduino/arduino-cli/releases/download/v0.35.0/0.35.0-checksums.txt
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_configuration.schema.json
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_32bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_64bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARM64.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARMv6.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Linux_ARMv7.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_macOS_64bit.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_macOS_ARM64.tar.gz
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_proto.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_32bit.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_64bit.msi
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/arduino-cli_0.35.0_Windows_64bit.zip
https://github.com/arduino/arduino-cli/releases/download/v0.35.0/CHANGELOG.md
Failed to install arduino-cli

### To reproduce

curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh > install_arduino.sh
chmod a+x install_arduino.sh
./install_arduino.sh

### Expected behavior

cli should be installed in ./bin

### Arduino CLI version

0.35.0

### Operating system

Linux

### Operating system version

ARM64

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

alessio-perugini, 2024-01-03:
We released a patch a couple of minutes ago. Do you mind trying again? :pray:

yerachw, 2024-01-03:
Running: ./install_arduino.sh 0.34.2 
does work!

yerachw, 2024-01-03:
Yes. The patch works!
Thanks

alessio-perugini, 2024-01-03:
Glad to hear that, and sorry about the inconvenience :pray:

closed 2024-01-03 by alessio-perugini