arduino/arduino-cli #2533: Confused by the `config init --dest-dir` functionality

pillo79, 2024-02-09:
### Describe the request

As I understand there are 3 ways to initialize the CLI config:
* `arduino-cli config init` will initialize the config in `~/.arduino15/arduino-cli.yaml` and use `~/.arduino15` as data store;
* `arduino-cli config init --dest-file cfg.yaml` will initialize the config in `cfg.yaml` and use `~/.arduino15` as data store;
* `arduino-cli config init --dest-dir cfg` will create a `cfg` folder, initialize the config in `cfg/arduino-cli.yaml` and STILL use `~/.arduino15` as data store.

The first way is the generic default. 
The second is a variant that allows you to have a different CLI config but share the package repos. 
The third works exactly like the second but creates an (otherwise unused) directory. I was really surprised by this as I thought it would initialize a fully separate CLI environment. 

What is the use case of the `config init --dest-dir` option? Can this be edited to automatically point `directories.data` and `directories.download` to the specified folder?

### Describe the current behavior

above

### Arduino CLI version

0.35.2

### Operating system

Linux

### Operating system version

--

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

cmaglie, 2024-07-30:
Hi @pillo79 

I think the original idea of `config init --dest-file cfg.yaml` was to match the global flag `--config-file cfg.yaml`.
BTW the same did not happen for the `config init --dest-dir cfg` that has no global flag counter part `--config-dir ...`.

Maybe a better strategy is to add a global `--config-dir my/config/dir` flag that does two things:
* sets the configuration file to `my/config/dir/arduino-cli.yaml`
* sets the **default** data dir to `my/config/dir`, so it will be used to store platforms and downloads (note that the directory will not be set in the configuration, but just taken as "default").

WDYT?

pillo79, 2024-07-30:
I don't think that is enough, since the generated file specifies these paths explicitly, and it would still override the default set by `--config-dir` later.

However, it would be fine if doing a `config init --dest-dir cfg --config-dir cfg` would result in the paths being "set to a custom default" and then written in `cfg/arduino-cli.yaml`.

pillo79, 2024-07-30:
Sorry didn't check the latest CLI behavior. I think this would be great! :smile: :+1:

closed 2024-08-02 by cmaglie