arduino/arduino-cli #2534: Compilation Error avr-g++ no such file or directory

Aterox-os, 2024-02-10:
### Describe the problem

Hello,
I know that this error was reported multiple times but none of the solutions of the other posts worked for me. I am getting the following error:
`Error during build: fork/exec /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: no such file or directory`

I tried to install it on a Raspberry Pi with a fresh Debian lite installation.

### To reproduce

I followed the official guide and this nonofficial guide. None of them worked.
https://www.caronteconsulting.com/en/news/how-to/raspberry-arduino-cli/

### Expected behavior

Compile my code

### Arduino CLI version

0.35.2

### Operating system

Linux

### Operating system version

Raspbian / Linux 12

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

matthijskooijman, 2024-02-13:
Does the file `/root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++` exist and is it executable?

What does `ldd /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++` output?

Aterox-os, 2024-02-17:
Sorry for the late response. I get not a dynamic executable.

matthijskooijman, 2024-02-18:
> I get not a dynamic executable.

Hm, I just tested running `ldd` on a raspberry pi on an x86_64 executable and I get the same, so maybe your av-g++ binary also has the wrong architecture.

IIUC, arduino-cli autodetects your architecture (or maybe has it hardcoded based on for which architecture the arduino-cli binary is built), and based on that downloads the right compiler tarball. It looks a bit like it might have downloaded the wrong one (but that's just a theory so far). I'm not quite sure how to get arduino-cli to print the arch it thinks you are on, though.

Some more commands to figure out more:

```
file  /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
file $(which arduino-cli)
ls ~/.arduino15/staging/packages/
```

closed 2024-02-19 by Aterox-os