arduino/arduino-cli #2535: why use program.pattern_args except upload.pattern_args when 'arduino_cli upload' with ESP32 ?

zbn10, 2024-02-12:
### Describe the problem

I have problem when executing (a)below with flesh ESP32S3 because it doesn't write bootloader.bin, partitions.bin and boot_app0.bin. ESP32 doesn't boot correctly.

When I examined Platform.txt, I found the following.
I think arduino-cli uses tools.esptool_py.program.pattern_args instead of upload.pattern_args.
Why ? Is it correct ?

tools.esptool_py.upload.pattern_args=--chip esp32s3 --port "{serial.port}" --baud 921600
  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 4MB
  0x0 "{build.path}/{build.project_name}.bootloader.bin"
  0x8000 "{build.path}/{build.project_name}.partitions.bin"
  0xe000 "C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/partitions/boot_app0.bin" 
  0x10000 "{build.path}/{build.project_name}.bin"

tools.esptool_py.program.pattern_args=--chip esp32s3 --port "{serial.port}" --baud 921600
  --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB
  0x10000 "{build.path}/{build.project_name}.bin"

### To reproduce

The (a)command results (b).
(a) arduino-cli.exe upload -v -t -b esp32:esp32:esp32s3:CDCOnBoot=cdc,USBMode=default,UploadMode=default -p COM5 --input-dir ./build/bin -P esptool .
(b) "C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32s3 --port "COM5" --baud 921600  --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x10000 "./build/bin/Project.ino.bin"

arduino core
esp32:esp32          2.0.14

### Expected behavior

Arduino IDE 2.2.1 upload procedure on the same machine results (c).
ESP32 works correctly.

(c) "C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32s3 --port "COM5" --baud 921600  --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 "HelloWorld.ino.bootloader.bin" 0x8000 "HelloWorld.ino.partitions.bin" 0xe000 "esp32\2.0.14/tools/partitions/boot_app0.bin" 0x10000 "HelloWorld.ino.bin" 

### Arduino CLI version

0.35.2

### Operating system

Windows

### Operating system version

windows11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-02-12:
Hi @zbn10.
> I think arduino-cli uses tools.esptool_py.program.pattern_args instead of upload.pattern_args.
Why?

Because you added the `-P` flag to your `arduino-cli upload` command. This causes Arduino CLI to perform [the `program` action](https://arduino.github.io/arduino-cli/dev/platform-specification/#upload-using-an-external-programmer) instead of [the `upload` action](https://arduino.github.io/arduino-cli/dev/platform-specification/#sketch-upload-configuration). You will get the same result if you select **Sketch > Upload Using Programmer** in Arduino IDE instead of doing a normal "Upload".

> Is it correct?

Yes. If you want the `upload` action, just remove the `-P` flag from your `arduino-cli upload` command.

closed 2024-02-12 by per1234

zbn10, 2024-02-12:
Thanks @per1234 
It's resolved as you mentioned.