arduino/arduino-cli #2581: arduino-cli monitor command does not use port in sketch.yaml file

marcelofern, 2024-04-12:
### Describe the request

The `arduino-cli monitor` command does not pick up the port written in the `sketch.yaml` file.
It would be great if it could, so that I can run `monitor` the same way I run other commands (compile/upload)

### Describe the current behavior

For example

```sh
$ ls                                                                                     
hello_world.ino  sketch.yaml

$ cat sketch.yaml                                                                        
default_fqbn: arduino:avr:uno
default_port: /dev/ttyUSB0

$ arduino-cli monitor                                                                    
Error: required flag(s) "port" not set
Usage:
  arduino-cli monitor [flags]

Examples:
  arduino-cli monitor -p /dev/ttyACM0
  arduino-cli monitor -p /dev/ttyACM0 --describe

Flags:
      --board-options strings        List of board options separated by commas. Or can be used multipl
e times for multiple options.
  -c, --config strings               Configure communication port settings. The format is <ID>=<value>
[,<ID>=<value>]...
      --describe                     Show all the settings of the communication port.
      --discovery-timeout duration   Max time to wait for port discovery, e.g.: 30s, 1m (default 1s)
  -b, --fqbn string                  Fully Qualified Board Name, e.g.: arduino:avr:uno
  -h, --help                         help for monitor
  -p, --port string                  Upload port address, e.g.: COM3 or /dev/ttyACM2
  -l, --protocol string              Upload port protocol, e.g: serial
  -q, --quiet                        Run in silent mode, show only monitor input and output.
      --raw                          Set terminal in raw mode (unbuffered).
      --timestamp                    Timestamp each incoming line.

Global Flags:
      --additional-urls strings   Comma-separated list of additional URLs for the Boards Manager.
      --config-file string        The custom config file (if not specified the default will be used).
      --format string             The output format for the logs, can be: text, json, jsonmini, yaml (
default "text")
      --log                       Print the logs on the standard output.
      --log-file string           Path to the file where logs will be written.
      --log-format string         The output format for the logs, can be: text, json
      --log-level string          Messages with this level and above will be logged. Valid levels are:
 trace, debug, info, warn, error, fatal, panic
      --no-color                  Disable colored output.

required flag(s) "port" not set
```

### Arduino CLI version

Version: 0.35.2-arch Commit: 01de174c7e3c08b9c4db121e77a5f7947f968097

### Operating system

Linux

### Operating system version

Arch linux, so it's rolling release sorry.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

alessio-perugini, 2024-04-15:
@marcelofern this was fixed in https://github.com/arduino/arduino-cli/pull/2329 But currently, it's only present in the master branch. It will be shipped with the next release which should be the `v1.0.0`

marcelofern, 2024-04-15:
Legend @alessio-perugini thank you!